OpenDroneMap/ODM #30: texturing_orthophoto_spotscale_additions branch. libboost1.48-all-dev install error on Ubuntu

mtrl, 2015-01-06:
On attempting to ./install.sh the texturing_orthophoto_spotscale_additions branch on Ubuntu 14.04 the apt-get install libboost1.48-all-dev line fails as it cannot be found in the apt repos.

Changing the apt-get install command to "libboost-all-dev" allows the install proceed. This installs version  "1.54.0.1ubuntu1" which may not be the correct one.

smathermather, 2015-01-07:
I'll test on known dataset with 14.04 tomorrow. Can you do a pull request against the branch?

dbaldwin, 2015-01-07:
I've confirmed the same issue. As @mtrl suggested, making the change to libboost-all-dev got me through the installation cleanly. This was on an EC2 c3.2xlarge instance running Ubuntu 14.04 64 bit.

smathermather, 2015-01-11:
Have you tested it on a dataset?

JesseCrocker, 2015-01-23:
I tested this on the "pacifica" dataset, and it worked properly.

smathermather, 2015-02-03:
Report here: https://github.com/OpenDroneMap/odm_vagrant/pull/4 from @bgirardot that install fails with libboost-all-dev possibly with 12.04. May need to out an if/fi in place to test Ubuntu version and install appropriately.

smathermather, 2015-02-05:
I'm gonna consider this one closed (again) :)

PR #42 (2015-01-23): Change libboost1.48-all-dev to libboost-all-dev.
Fixes #30
closed 2015-02-05 by smathermather