OpenDroneMap/ODM #32: Error cmaking odm_texturing

dbaldwin, 2015-01-08:
When trying to install the orthophoto branch I get the following related to cmaking odm_texturing:

CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/OpenDroneMap/src/odm_texturing/CMakeFiles/CMakeOutput.log".

I've been trying to get to the bottom of this but haven't had much luck. Any insight would be much appreciated.

dbaldwin, 2015-01-08:
I just fired up another Ubuntu 14.04 VM (Amazon EC2) to make sure it wasn't a one-off issue. I ran into the same error. What I did to get this working is:

sudo apt-get install libopencv-dev

And bam! I was able to install. How do the opencv dependencies get installed in the first place? I can explicitly add this package and submit a pull request, but not sure if that's the right thing to do.

smathermather, 2015-01-11:
Add that, and I'll test on 12.04...

smathermather, 2015-01-12:
@dbaldwin -- can you do a pull request for this change. Go ahead and add it to the install script, test on your end, and I'll test here.

closed 2015-01-20 by smathermather