OpenDroneMap/ODM #36: Installation on EC2

soler2000, 2015-01-15:
When I try to install this on Amazons EC2 ubuntu server,  the installation halts at   

> cmvs/pmvs
    - cleaning cmvs
    - building cmvs

then goes no further

dbaldwin, 2015-01-16:
I'm assuming you're on Ubuntu 14.04 64 bit since that's one of the more common EC2 images. I've run ODM on there at least a dozen times and have run into issues here and there. It's hard to say without more info. In the ODM directory check the logs directory and look for cmvs_2_build.log since that appears to be where it's hanging up. Perhaps that will have some useful info. If you can't figure it out then post the log contents here and if you'd be up for setting up a user account on the EC2 instance I'll be glad to login and take a look.

soler2000, 2015-01-16:
Yes,  I am running Ubuntu 14.04 64,  from the log it looks like it ran out of virtual memory,   could you provide an example of the specs that are min and recommended for this application?

Please see the log below

g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include pmvs2.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/image/camera.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/image/image.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/image/photo.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/image/photoSetS.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/detectFeatures.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/detector.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/dog.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/expand.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/filter.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/findMatch.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/harris.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/optim.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/option.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/patch.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/patchOrganizerS.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/point.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/pmvs/seed.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include ../base/numeric/mylapack.cc
g++ -c -O2 -Wall -Wno-deprecated -DNUMBITS=64 -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include -I/home/ubuntu/git/OpenDroneMap/include/metisLib/ -fopenmp -DNUMBITS=64  cmvs.cc
g++ -c -O2 -Wall -Wno-deprecated -DNUMBITS=64 -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include -I/home/ubuntu/git/OpenDroneMap/include/metisLib/ -fopenmp -DNUMBITS=64  ../base/cmvs/bundle.cc
g++ -c -O2 -Wall -Wno-deprecated -DNUMBITS=64 -I/usr/local/include -I/home/ubuntu/git/OpenDroneMap/include -I/home/ubuntu/git/OpenDroneMap/include/metisLib/ -fopenmp -DNUMBITS=64  ../base/cmvs/graclus.cc
g++ -o genOption genOption.cc
virtual memory exhausted: Cannot allocate memory
make: *** [cmvs.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [bundle.o] Error 1
../base/pmvs/optim.cc: In static member function ‘static double PMVS3::Coptim::my_f_ssd(const gsl_vector*, void*)’:
../base/pmvs/optim.cc:762:9: warning: variable ‘flag’ set but not used [-Wunused-but-set-variable]
     int flag;

dbaldwin, 2015-01-16:
The minimum I've run ODM with is 16 GB memory. I will fire up an 8 GB instance in a bit and see if I run into the same problem. What are you running it on?

dbaldwin, 2015-01-16:
I just launched a c4.xlarge instance on EC2 (8GB memory) and installed in about 2 minutes with no problems. Are you cloning and installing the master branch or another?

soler2000, 2015-01-16:
It was because I opened it on the basic free option,  after trying it on a larger installation everything was fine

closed 2015-01-16 by soler2000