OpenDroneMap/ODM #39: Help with odm_texturing

dbaldwin, 2015-01-23:
@smathermather I could use a little help understanding the odm_texturing process. Everything runs fine, but the results look a bit scattered. Or maybe I don't understand what to do next with the various textured outputs. I'd like to put together a YouTube video giving an overview of this feature if you can help me understand. Here is the output:

https://www.dropbox.com/sh/2pvvgtwacn2ifrf/AACSkzEzezGC-mQJQSdTraDFa?dl=0

Also, keep in mind this is the same dataset I ran in a previous video and the textured mesh turned out great. Any insight would be much appreciated.

MultiD, 2015-01-23:
Had a similar issue with odm_texturing process with another dataset of mine and it looks odd same as yours.

smathermather, 2015-01-23:
Hi @dbaldwin and @MultiD,

Good news-- that is what those files are supposed to look like. Import the associated mesh into meshlab (_odm_textured_model.obj_), and those textures should display in place on the mesh.

Thanks for taking this on!

closed 2015-01-23 by smathermather

smathermather, 2015-01-23:
Added this info to the Readme https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/README.md .

I haven't documented orthophoto creation yet, but orthophoto creation will get a lot simpler in the next couple weeks, so I'm holding off until those new updates are in.

dbaldwin, 2015-01-23:
@smathermather It doesn't get much easier than that. Thanks for clueing us in.

smathermather, 2015-01-23:
My pleasure, @dbaldwin.

smathermather, 2015-01-26:
@dbaldwin -- are you willing to add the raw dataset to https://github.com/OpenDroneMap/odm_data , perhaps under CC0 or CC-by-Attribution license?

dbaldwin, 2015-01-27:
@smathermather Yes most definitely. I will submit a pull request in the next day or two. I also have some new imagery I'm working on for another property and will be happy to share that as well....after I run it through ODM of course.