OpenDroneMap/ODM #47: software does not recognize my camera go pro?

paradigma-x, 2015-02-04:
Hello,
I would ask how to solve this problem: 

no CCD width or focal length found for G0035260.JPG - camera: " "sh: 1: jhead: not found
sh: 1: jhead: not found
sh: 1: jhead: not found
sh: 1: jhead: not found
sh: 1: jhead: not found

    no CCD width or focal length found for G0035261.JPG - camera: " "

    found no usable images - quitting
Died at /home/innova/OpenDroneMap/run.pl line 426.
I tried to insert the value of the camera, as advertised:
.........
So, we'll add the following line to our ccd_defs.pl:

 "SONY DSC-HX5V"                            => 6.104,  # 1/2.4"

And so others can use it, we'll do a pull request to add it to our array for everyone else.

my ccd:
"GOPRO BLACK Edition"                    => 6.17, #1/2.3"                                                                              
but it did not work, any idea that does not recognize the information loaded?

regards

paradigma-x, 2015-02-04:
thas is my file ccd_defs.pl

"SONY DSC-W1"                              => 7.176,  # 1/1.8"
     "SONY DSC-W30"                             => 5.760,  # 1/2.5"
     "SONY DSC-W50"                             => 5.75,   # 1/2.5"
     "SONY DSC-W5"                              => 7.176,  # 1/1.8"
     "SONY DSC-W7"                              => 7.176,  # 1/1.8"
     "SONY DSC-W80"                             => 5.75,   # 1/2.5"
     "SONY DSC-HX5V"                            => 6.104,  # 1/2.4"
     "SONY NEX-7"                               => 23.5, # APS-C" 
     "SONY A7"                                  => 35.9, # Full-frame
     "SONY A7R"                                    => 35.9, # Full-frame
     "GOPRO HERO 3"                             => 6.17,    # 1/2.3"

any idea?

dbaldwin, 2015-02-04:
It looks like the run script can't find jhead which is used to pull the exif data from the image. There is where it gets installed:

https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/install.sh

It looks to me that somehow it didn't get installed on your system. What version of Ubuntu are you running and were you able to successfully install ODM. Meaning did you get all the way to the end of the install script? You would know this by seening "script finished - date & time" at end as indicated here:

https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/install.sh#L476

Also, I'm not sure how ODM is going to handle the GoPro lens distortion. This could cause problems. If it does you can always "flatten" your images as mentioned here:

http://publiclab.org/notes/huslage/05-01-2013/resolving-barrel-distortion-on-a-gopro-hero3

I have a set of GoPro Hero 3+ images that I'll run through ODM today and share my results. Also, based on that article above can you confirm whether or not you have lens info in your EXIF data? The article seems to suggest there isn't any for the 3, but my 3+ does have it. So I'm curious to know if there's a difference.

wildintellect, 2015-02-08:
I have a script for undistorting GoPros
https://github.com/wildintellect/lenscorrection

You will need a Lensfun lens model though fro the Hero3, my tests were with a Hero2. I have blog post describing the whole process.
http://blog.wildintellect.com/blog/lenscorrect

I have not tested the results with ODM yet. I suspect it will fail because the camera model can't be detected, and I'm not sure what parameters to use post-distortion correction.

wildintellect, 2015-02-08:
Ah, one thing I did notice about the Exif data on GoPros (at least Hero2) is that it has a camera model but not a lens model. So you might have to manually override that in ODM somehow.

BangorSkis, 2015-02-26:
After first running the images through Adobe Camera Raw with the lens correction, for the GoPro Hero 3 Black Edition, I've been adding a line with:

```
  "GoPro Hero3-Black Edition": 6.16,
```

As that's the specific characters that the EXIF data in my images has for the GP3-Black.

First results out of ODM at lowered resolution (2400px) look promising. Firing up a more powerful machine to run at full resolution.....

smathermather, 2015-02-26:
Awesome!

BTW, if you get a chance to test it, it'd be cool to know how well this solution works for correcting for distortion for GoPros and such:

http://www.janeriksolem.net/2014/05/how-to-calibrate-camera-with-opencv-and.html

smathermather, 2015-02-26:
Hehe, just saw @wildintellect's post above. This is better. If we hack away at this at FOSS4GNA (just sent an e-mail) we might also include vignetting corrections, if possible... .

cgdnorth, 2015-03-09:
Back on topic, I am getting the same error as @paradigma-x, is there anything I can do to fix this. I am using Ubuntu 14.04 on VirtualBox running on a Mac.

nirabpudasaini, 2015-03-10:
I did some lens correction using GIMP. I found that the images after Lens
correction were missing EXIF tags and i had to copy them from the original
images. Please check that is not the case with you.

cgdnorth, 2015-03-10:
I am getting this error on original images as well as the ones I did lens correction on in Lightroom.

smathermather, 2015-03-11:
@cgdnorth -- I don't know if anyone has a solution yet for GoPros and other cameras without a lens model, but this is a good one for the users list:

http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users

Having a solution is high priority.

wildintellect, 2015-03-12:
I'm working on the solution. I'm adding some code today to my https://github.com/wildintellect/lenscorrection code that will write new exif to the corrected photos. From there I'll add a line to the ccc_defs.json file with the new corrected Hero2 sensor size (~5mm after correction). This should allow us to at least try running the data.

smathermather, 2015-11-23:
Lens correction makes a big difference. Using http://www.janeriksolem.net/2014/05/how-to-calibrate-camera-with-opencv-and.html code works pretty well with GoPro according to some initial tests by @dakotabenjamin, but I he hasn't put it through ODM yet.

@jesolem -- any standard way to estimate chip size after un-distorting an image? AFAIK, this is all we'd need in addition to your script to get wide angle stuff to work in ODM.

smathermather, 2018-03-23:
See also: https://github.com/OpenDroneMap/CameraCalibration

closed 2018-03-23 by smathermather