OpenDroneMap/ODM #48: Can't locate JSON.pm

mtrl, 2015-02-06:
I'm seeing the following error when running "run.pl" against a project after installing.

```Can't locate JSON.pm in @INC (@INC contains: /etc/perl /usr/local/lib/perl/5.14.2  /usr/local/share/perl/5.14.2 /usr/lib/perl5 /usr/share/perl5 /usr/lib/perl/5.14 /usr/share/perl/5.14 /usr/local/lib/site_perl .) at /home/vagrant/OpenDroneMap/run.pl line 16.
BEGIN failed--compilation aborted at /home/vagrant/OpenDroneMap/run.pl line 16.```

I'm seeing this both on the standard vagrant VM and on other 12.04 VMs.

`locate JSON`
gives me the following
```/usr/share/perl/5.14.2/JSON
/usr/share/perl/5.14.2/JSON/PP
/usr/share/perl/5.14.2/JSON/PP.pm
/usr/share/perl/5.14.2/JSON/PP/Boolean.pm```

mtrl, 2015-02-06:
I resolved the problem with
`sudo apt-get install libjson-perl`

bgirardot, 2015-02-06:
It looks like  libjson-perl isn't installed in if install.sh script detects 12.04.

smathermather, 2015-02-07:
Fixed:

https://github.com/OpenDroneMap/OpenDroneMap/commit/0f178c2475e1c852c4ac3c19b3f3e15a5dadba8d

closed 2015-02-07 by smathermather