OpenDroneMap/ODM #51: Create a mailing list

wildintellect, 2015-02-08:
Seems that the Issue tracker might get full of posts that aren't necessarily bugs. Things like discussing tuning parameters, or troubleshooting failed installations (not code errors). In my example, I actually wanted to post about a failure to build the Kermit data set correctly (output image is distorted). This isn't necessarily a bug yet.

I'd recommend getting an OSGeo hosted mailing list. Simply requires a request to the OSGeo System Administration Committee. I could file the request if thats what's wanted. 
http://wiki.osgeo.org/wiki/SAC:Mailing_Lists#Requesting_a_new_list

Google groups is another option, but is a pain for non-google users.

smathermather, 2015-02-09:
Awesome. Have at it!

wildintellect, 2015-02-09:
Do we want just 1 opendronemap@osgeo, or a split odm-users, and odm-dev?

smathermather, 2015-02-09:
I guess split 'em for long-term sake.

wildintellect, 2015-02-09:
Request made https://trac.osgeo.org/osgeo/ticket/1469 will post the links to join when they are ready.

smathermather, 2015-02-09:
Awesome. Thank you.

wildintellect, 2015-02-09:
Sign ups are open. We should add these to the main website:
http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users
http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-dev

 Archives category at:
 http://osgeo-org.1560.x6.nabble.com/OpenDroneMap-f5186282.html

closed 2015-02-09 by wildintellect

smathermather, 2015-02-09:
w00t! Done: http://opendronemap.github.io/odm/pages/contact.html