OpenDroneMap/ODM #52: OpenDroneMapMosaic Production

majorpr13, 2015-02-13:
Is OpenDroneMap capable of producing mosaics? I have processed several sets of images and have not seen anything like this in the output files. If it does not, perhaps add this to a feature request.
Thanks.

smathermather, 2015-02-13:
It is capable of producing mosaics. I'll work on some documentation this weekend.

majorpr13, 2015-02-25:
Was there any luck on this? I am interested in seeing if I can produce a mosaic correction without running through the 3D reconstruction process? Also, does any of the reconstruction process take into account aircraft attitude?

smathermather, 2015-02-25:
Sorry, I have not gotten to this yet.

A mosaic is dependent upon 3D reconstruction, as that is how the relationship between the photos is established and the 3D reconstruction also allows for the use of a digital surface model for orthorectification. This gives superior results to simple matching.

I may have more time for documentation this week. Sorry for the delay.

majorpr13, 2015-02-25:
Ok last question...I probably should be pushing these through the email listserv....Has there been any code developed or developing to reduce the bundler matching process...For example since we may have the GPS locations of the images I could determine the relevant neighboring images to perform Bundler PMVS or whatever on....This way we can reduce the the search from n^2 which I think is what it is doing?

smathermather, 2015-02-25:
Yes, it will be a major improvement when we optimize the 'match' step. Integration of OpenSfM may help with this (https://github.com/OpenDroneMap/OpenDroneMap/pull/50), but that's not quite road ready, and would require some optimization itself for the matching problem.

It might be a good solution to integrate e.g. [SFCGAL](http://www.sfcgal.org/) and prebuild 3D intersections a la:

https://smathermather.wordpress.com/2013/12/15/uas-drone-footprint-geometries-calculated-in-postgis-with-sfcgal-for-real-this-time/

There is a milestone/issue for this improvement:
https://github.com/OpenDroneMap/OpenDroneMap/issues/14

Also, yes the list is better for this. See a similar conversation here:
http://lists.osgeo.org/pipermail/opendronemap-users/2015-February/000006.html

Cheers

smathermather, 2015-11-23:
Closed with https://github.com/OpenDroneMap/OpenDroneMap/pull/203. Cheers.

closed 2015-11-23 by smathermather