OpenDroneMap/ODM #53: Georeferencing Error

majorpr13, 2015-02-13:
All,
When running OpenDroneMap I get an abort error thrown during the georeferencing step. What does this error code mean? See the below for output. I am running a 64bit Ubuntu system.

  - running georeferencing - 

quitting cause: 
	"/home/scoriolinux/Desktop/VTImages/Left/../../OpenDroneMap/bin/odm_georef" -bundleFile /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/pmvs/bundle.rd.out -gcpFile /home/scoriolinux/Desktop/VTImages/Left/gcp_list.txt -imagesPath /home/scoriolinux/Desktop/VTImages/Left/ -imagesListPath /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/pmvs/list.rd.txt -bundleResizedTo 1200 -inputFile /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200-results/odm_texturing/odm_textured_model.obj -outputFile /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200-results/odm_texturing/odm_textured_model_geo.obj -logFile /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/odm_georeferencing/odm_georeferencing_log.txt

dbaldwin, 2015-02-13:
Check your log file located here:

/home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/odm_georeferencing/odm_georeferencing_log.txt

You'll likely see the error near the bottom of this file. Let us know and we can take it from there.

majorpr13, 2015-02-13:
Thanks. The error from the log file can be seen below. 

/home/scoriolinux/Desktop/VTImages/Left/../../OpenDroneMap/bin/odm_georef_log.txt
Arguments given
-bundleFile
/home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/pmvs/bundle.rd.out
-gcpFile
/home/scoriolinux/Desktop/VTImages/Left/gcp_list.txt
-imagesPath
/home/scoriolinux/Desktop/VTImages/Left/
-imagesListPath
/home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/pmvs/list.rd.txt
-bundleResizedTo
1200
-inputFile
/home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200-results/odm_texturing/odm_textured_model.obj
-outputFile
/home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200-results/odm_texturing/odm_textured_model_geo.obj
-logFile
/home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/odm_georeferencing/odm_georeferencing_log.txt

Reading cameras from: /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/pmvs/bundle.rd.out
Reading GCPs from: /home/scoriolinux/Desktop/VTImages/Left/gcp_list.txt
Images location is set to: /home/scoriolinux/Desktop/VTImages/Left/
Reading image list from: /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/pmvs/list.rd.txt
Bundle resize value is set to: 1200
Reading textured mesh from: /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200-results/odm_texturing/odm_textured_model.obj
Writing output to: /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200-results/odm_texturing/odm_textured_model_geo.obj
Log file path was set to: /home/scoriolinux/Desktop/VTImages/Left/reconstruction-with-image-size-1200/odm_georeferencing/odm_georeferencing_log.txt
Error in Georef:
Failed opening /home/scoriolinux/Desktop/VTImages/Left/gcp_list.txt for reading.

majorpr13, 2015-02-13:
There is however no gcp_list.txt file in that address....

smathermather, 2015-02-13:
Hi @majorpr13 -- I have some code to post yet that will exit gracefully when there's no gcp's.

majorpr13, 2015-02-14:
Excellent. Is there an example of how to generate a ground control point file or what one would look like if properly formatted that the algorithm could execute on? I looked through the sample files and dont believe that I saw one present. 
Thanks.

smathermather, 2015-02-14:
Sure is: https://smathermather.wordpress.com/2015/02/04/opendronemap-orthophotos-and-gcps/

This could be added to the Seneca dataset as a pull request, as those coords are for Seneca (just haven't gotten to it yet).

w0rcy, 2015-02-15:
@Smathermather, it looks like targets 5 and 7 are the same. Is this correct?

smathermather, 2015-02-15:
Ignore the Coal Oil Point Reserve info -- that GCP file is actually for Seneca. I just conflated the two for the blog post because the copr data is so nicely illustrated.

That said, I think you are right: 7 is supposed to be pointing to the lower right corner. Pinging @wildintellect to confirm, referencing https://github.com/OpenDroneMap/odm_data_copr/blob/master/COPR_GPS_Points.pdf .

wildintellect, 2015-02-15:
I'll check with @MicheleTobias (just sent a team invite), and get a corrected pdf. But yes it looks like 7 in the diagram is incorrect. FYI those patterns are based on flag semaphore http://en.wikipedia.org/wiki/Flag_semaphore

MicheleTobias, 2015-02-15:
Yup, there's an error.  If you zoom in on point 7 in the photo at the top, you can see that the white bar should point to the lower right corner and not be the same as point 5.  I will fix that and upload a fixed diagram.

smathermather, 2015-02-15:
Awesome.

w0rcy, 2015-02-16:
Does the software recognize the markers, or is that for the user's reference? If so, do they have to be white on black, or can they be black on white?

smathermather, 2015-02-16:
The software does not recognize the markers as GCPs -- it is in specifying the image coordinates and geo-coordinates that the georeferencing can take place. In short, they can be any color that you can recognize.

w0rcy, 2015-02-17:
I see in the code that the correspondence file must be called gcp_list.txt, but is it possible to use x,y,z decimal coordinates from gps, rather than the UTM coordinates the way they are in the example correspondence file? How would that header be specified?

smathermather, 2015-02-17:
The header is free-form, so it doesn't matter from a file format standpoint

That said, if you used geographic coordinates from GPS, your pixel sizes will be in geographic coordinates, which may not make any sense, and would distort things.

We should start moving these conversations to the mailing lists:

Users' mailing list: http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users

Developer's mailing list: http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-dev

okiave, 2015-03-04:
I ran into this same error. Just curious,
would having the GPS data stamped into images EXIF 
help clear this error?

I understand that's not what GCP is, but the process does take into
account the LAT LONGS.

smathermather, 2015-03-04:
Having GPS in the exif does circumvent the error. I also have a patch, so this doesn't error out if the gcp file doesn't exist. I will commit it tonight.

smathermather, 2015-03-06:
Having EXIF would avoid the error. So would using this branch:

https://github.com/OpenDroneMap/OpenDroneMap/tree/graceful-exit-gcp

Can you test this branch without EXIF and without a GCP text and report
back?

Thanks!

okiave, 2015-03-06:
That's odd. I can't clone, pull, fetch or add that branch.
Any ideas?

smathermather, 2015-03-08:
Try something like this:

https://www.quora.com/Git-revision-control/How-do-I-clone-this-branch-that-hasnt-been-merged-into-master-in-GitHub?share=1

Best,
Steve

nirabpudasaini, 2015-03-08:
I see that there is an option --odm_georeferencing-useGcp that can be set to false to skip reading from the GCP file.

/OpenDroneMap/run.pl --odm_georeferencing-useGcp "false" this command sould skip reading GCP and use the EXIF to generate the cordList.txt for georefrencing.

However the UtmExtractor is not working correctly for me. It fails to find the Latitude and throws exception despite my images having all the data. This is the output when i rung jhead on the image file.

File name    : DJI02089.jpg
File size    : 11324087 bytes
File date    : 2015:03:08 10:58:31
Camera make  : DJI
Camera model : PHANTOM VISION FC200
Date/Time    : 2015:02:05 10:51:17
Resolution   : 4608 x 3456
Focal length :  5.0mm
Exposure time: 0.0006 s  (1/1667)
Aperture     : f/2.8
ISO equiv.   : 100
Whitebalance : Auto
Metering Mode: center weight
Exposure     : program (auto)
Jpeg process : Progressive
GPS Latitude : N 27d 43m 33.94s
GPS Longitude: E 85d 20m 33.11s
GPS Altitude :  1352.30m
JPEG Quality : 100

You can see that there is Latitude Longitude and Altitude but still the UtmExtractor exist with log entry

Error in OdmExtractUtm:
Image is missing GPS Latitude data

smathermather, 2015-03-08:
@nirabpudasaini -- can you open a separate issue for this? Also, are your data shareable so I can do some testing?

okiave, 2015-03-08:
@smathermather I got the same error specific to the missing georef txt file.
Now it could just be I downloaded the wrong branch. This is the command I used to 
download the branch you specified.

>$ git clone https://github.com/OpenDroneMap/OpenDroneMap.git -b graceful-exit-gcp --depth 1

smathermather, 2015-03-09:
Is this error specific to that branch, or do you see it also on the main
branch (gh-pages)?

nirabpudasaini, 2015-03-09:
Open a sperate issue. My data is available at https://github.com/nirabpudasaini/mission01.

okiave, 2015-03-10:
The same error code 256 shows up on both branches. 
I just ran it again to make sure. Here's the log it references if
your curious. 

https://www.dropbox.com/s/xcq6jybytrlnupg/odm_georeferencing_log.txt?dl=0

>  - running georeferencing - Tue Mar 10 09:17:37 UTC 2015

>libdc1394 error: Failed to initialize libdc1394

>quitting cause:
>	"/vagrant/OpenDroneMap/bin/odm_georef" -bundleFile /vagrant/klatt/reconstruction-with-image->size-2400/pmvs/bundle.rd.out -gcpFile /vagrant/klatt/gcp_list.txt -imagesPath /vagrant/klatt/ ->imagesListPath /vagrant/klatt/reconstruction-with-image-size-2400/pmvs/list.rd.txt -bundleResizedTo >2400 -inputFile /vagrant/klatt/reconstruction-with-image-size-2400->results/odm_texturing/odm_textured_model.obj -outputFile /vagrant/klatt/reconstruction-with-image->size-2400-results/odm_texturing/odm_textured_model_geo.obj -logFile /vagrant/klatt/reconstruction->with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt
>returned with code 256

okiave, 2015-03-11:
So upon further testing it seems I have some sort of weird bug.
I did a test with the <a href="https://github.com/OpenDroneMap/odm_data_caliterra.git">geocoded Caliterra imagery</a> by @dbaldwin and got the exact
same 256 error using both branches. 

I'm running Ubuntu 14.04.1 LTS, via vagrant 7 vbox.

dbaldwin, 2015-03-12:
> I did a test with the geocoded Caliterra imagery

FWIW I ran the dataset this AM to do some testing with my recently added
GCPs. While the georeferenced output is still a work in progress I did get
a nice orthophoto. You can see it here:

https://www.dropbox.com/s/42csbbk0dklskmp/odm_orthphoto.png?dl=0

This was run with Ubuntu 14.04 LTS on Amazon EC2. I ran into so many
problems with Vagrant and VirtualBox that I finally gave up. I do plan on
revisiting but for the time being I'll spend the couple of dollars to run
this on Amazon.

smathermather, 2015-03-16:
Hey @dbaldwin -- that looks good! Can you add the gcp file to the data repo so others can use it as an example?

dbaldwin, 2015-03-21:
@smathermather Sure thing. I'll get it added in the next couple of days after I do some more testing.

closed 2016-10-12 by dakotabenjamin