OpenDroneMap/ODM #54: ODM_Texturing: Image not found

merkato, 2015-02-14:
I have problem like this:

Bundle path was set to: /home/mechanik/dev/SfM/reconstruction-with-image-size-1200/pmvs/bundle.rd.out
Images path was set to: /home/mechanik/dev/SfM/
Images list path was set to: /home/mechanik/dev/SfM/reconstruction-with-image-size-1200/pmvs/list.rd.txt
Input model path was set to: /home/mechanik/dev/SfM/reconstruction-with-image-size-1200-results/odm_mesh-0000.ply
Output folder path was set to: /home/mechanik/dev/SfM/reconstruction-with-image-size-1200-results/odm_texturing/
The texture resolution was set to: 4096
The resized resolution used in bundler was set to: 1200
The resolution to texture with was set to: 3600
Log file path was set to: /home/mechanik/dev/SfM/reconstruction-with-image-size-1200/odm_texturing/odm_texturing_log.txt
Successfully loaded 135621 polygons from file.
Successfully read the bundle file.
Successfully read the image list file.
Error in OdmTexturing:
Error in OdmTexturing:
Failed to read image:
'/home/mechanik/dev/SfM//_MG_7067.jpg'

bgirardot, 2015-02-14:
Hi, please read through issue #33 comments, this sounds like it might be the uppercase file extension issue and that thread has a simple command line to rename the files to lowercase. Do your source files have uppercase file name extensions?

merkato, 2015-02-14:
Yes, It's related. I've renamed files to lowercase (name + extension), and odm_texturing is probably working ;) Right now it doesn't exit with error. Thanks for very fast resolve :)

closed 2015-02-14 by merkato