OpenDroneMap/ODM #55: odm_meshing problem

merkato, 2015-02-16:
I've was testing ODM with some photos sets. Workflow passed CMVS/PMVS stage correctly, generating option-0000.ply, that's very good quality. But on odm_meshing it throws "Segmentation fault (Core  Dumped)" error. When I've tried to start odm_meshing from terminal, that what I've get. 

odm_meshing_log.txt
Vertex count was manually set to: 0
Reading point cloud at: /home/en76-008/dev/boruszyn/reconstruction-with-image-size-1200-results/option-0000.ply
Successfully loaded 149118 points with corresponding normals from file.
Octree depth used for reconstruction is: 8
Estimated initial vertex count: 65536

Reconstruction complete:
Vertex count: 52569
Triangle count: 104955

Vertex count not specified, decimation cancelled.
Saving mesh to file.
Error in OdmMeshing:
Error when saving mesh to file:

How to eventually find the problem? 
With regards, Tomasz

smathermather, 2015-02-16:
Hi Tomasz,

Can you post the data anywhere for testing?

merkato, 2015-02-16:
Hi, 

I'm trying it on another computer, because in meantime I have OpenCV errors on odm_texturing stage with smaller part of this dataset. 

I will upload dataset as quick as possible.
Thanks for feedback.

merkato, 2015-02-16:
Bundle path was set to: /home/mechanik/dev/boruszyn/reconstruction-with-image-size-1200/pmvs/bundle.rd.out
Images path was set to: /home/mechanik/dev/boruszyn/
Images list path was set to: /home/mechanik/dev/boruszyn/reconstruction-with-image-size-1200/pmvs/list.rd.txt
Input model path was set to: /home/mechanik/dev/boruszyn/reconstruction-with-image-size-1200-results/odm_mesh-0000.ply
Output folder path was set to: /home/mechanik/dev/boruszyn/reconstruction-with-image-size-1200-results/odm_texturing/
The texture resolution was set to: 4096
The resized resolution used in bundler was set to: 1200
The resolution to texture with was set to: 3600
Log file path was set to: /home/mechanik/dev/boruszyn/reconstruction-with-image-size-1200/odm_texturing/odm_texturing_log.txt
Successfully loaded 199857 polygons from file.
Successfully read the bundle file.
Successfully read the image list file.
Visible faces: 193948
Occluded faces: 5909
Faces sorted into 7 textures.
Error in OdmTexturing:
/build/buildd/opencv-2.4.8+dfsg1/modules/core/src/matrix.cpp:323: error: (-215) 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows in function Mat
------------------------------------------------
Dataset is here: https://www.dropbox.com/s/cfso3kakx1b8o8w/boruszyn.zip?dl=0
And gcp_list.txt "helper" is here: https://www.dropbox.com/s/gqfvizeclg95f9o/boruszyn.pdf?dl=0

smathermather, 2015-02-16:
Is that the full dataset, or a subset?

merkato, 2015-02-17:
Hi Stephen,

It is a subset, and this subset throws this OpenCV error on two installations of ODM (both Ubuntu 14.04, one was used earlier for VisualSFM building - I was thinking, that maybe some libraries used there are not compatibile). 
I will upload full dataset later today.

Thanks very much for so great features in ODM :)

smathermather, 2015-02-18:
Cool. The subset may be _too_ subset. Looking forward to testing with the full dataset.

Cheers!

merkato, 2015-02-18:
https://www.dropbox.com/s/ey9sf98qcndgo8b/boruszyn_results.zip?dl=0

That's the full dataset. On my desktop computer (Core2Duo, 3GB RAM, Xubuntu 14.04 64b) it throws thst ODM_Meshing Segmentation Fault. And on laptop (same config), there's OpenCV error, on full dataset, or on subset.

smathermather, 2015-02-18:
Ok, running a test now. Will check on it in the AM.

merkato, 2015-02-18:
Heh, 
I have 'AM' only for next 2 hours ;) In Poland there is 09:50 AM. That's why I answer late night for you ;)

Cheers!

smathermather, 2015-02-18:
Haha! Well it's the AM here again. The process ran fine for me, but I threw it on a machine with 32GB of RAM, so perhaps you were running out of RAM on your laptop. I can zip up and send you the output if you like.

smathermather, 2015-02-18:
FYI, this is one of the best results I've seen yet. Would you be willing to share this dataset as part of https://github.com/OpenDroneMap/odm_data ? I'd love to include it:

[image]

[image]

merkato, 2015-02-18:
Ok, I will do pull request for that data. If you can send me an zip with results, I will be appreciated. On maillist I was asking about georeferencing problem with orthophoto. Can you check that? Maybe I'm doing something wrong way.

With regards,

smathermather, 2015-02-18:
Your output data are here: https://drive.google.com/file/d/0B5m8jMjoHnYlMzlYSENhakNOZUE/view?usp=sharing

I can't replicate, so I'm going to close this for now. Let me know if you think it should be re-opened.

And... I have now answered the first question on the mailing list!

closed 2015-02-18 by smathermather