OpenDroneMap/ODM #56: Error in OdmTexturing:

jhiner1978, 2015-02-16:
Im using the test files to do a test of this app, so far I like what I see but when I run it on my VM Ubuntu 64bit I get the same issue no matter which data folder I run it in. Here are the lines leading up the error

    ---- Total: 0 secs ----

      - running meshing - 

      - running texturing - 
    Error in OdmTexturing:
    Error in OdmTexturing:
    Failed to read image:
    '/home/jamison/Desktop/copr//IMG_0028.jpg'

    For more detailed information, see log file.

    quitting cause: 
	    "/home/jamison/Desktop/copr/../odm/bin/odm_texturing" -bundleFile /home/jamison/Desktop/copr/reconstruction-with-image-size-1200/pmvs/bundle.rd.out -imagesPath /home/jamison/Desktop/copr/ -imagesListPath /home/jamison/Desktop/copr/reconstruction-with-image-size-1200/pmvs/list.rd.txt -inputModelPath /home/jamison/Desktop/copr/reconstruction-with-image-size-1200-results/odm_mesh-0000.ply -outputFolder /home/jamison/Desktop/copr/reconstruction-with-image-size-1200-results/odm_texturing/ -textureResolution 4096 -bundleResizedTo 1200 -textureWithSize 3600 -logFile /home/jamison/Desktop/copr/reconstruction-with-image-size-1200/odm_texturing/odm_texturing_log.txt
    returned with code 256
    jamison@jamison-VirtualBox:~/Desktop/copr$

my guess is that it has something to do the fact that its not finding the images 

    Failed to read image:
    '/home/jamison/Desktop/copr//IMG_0028.jpg'

Need to figue out how to take out that extra '/' in the file address

bgirardot, 2015-02-16:
Hi, this is a known issue at the moment.
It is because the file name extensions are uppercase in the test images.
A quick fix is to cd into the directory of images and run this command:

```
rename 'y/A-Z/a-z/' *
```
and that will make the whole file name lowercase, things should run fine after that.

jhiner1978, 2015-02-16:
Thanks i'll give it a go... I would like to contribute where I can, I do web design and development so if you guys need help in that realm then I could help there

smathermather, 2015-02-17:
Working on fixing this in the example data in advance of a patch to compensate for the difference in naming.

smathermather, 2015-02-17:
OK. All example datasets have been fixed. We will still want a catch in the code for this, but at least now we won't immediately drive new users crazy... :

https://github.com/OpenDroneMap/odm_data_apt/commit/0ece0bfb43b653d87323a1f5f3c401bd94f87d7b
https://github.com/OpenDroneMap/odm_data_benchmark/commit/2ccef98ac0201eceec3cba206c63964fc6a97d4c
https://github.com/OpenDroneMap/odm_data_caliterra/commit/cb3476a538585ebff92af36329b4d92d49aeefcf
https://github.com/OpenDroneMap/odm_data_copr/commit/faa075331fe70cf88f11a4187fc719e75cebc8e8
https://github.com/OpenDroneMap/odm_data_glam/commit/82e18faf717de0e734ad82287330ebba840426c0
https://github.com/OpenDroneMap/odm_data_langley/commit/b28267b1eb396ba3802458ee770984df1ef1c0b9
https://github.com/OpenDroneMap/odm_data_pacifica/commit/76ab267840bb8867c77f9acb6921438ab8b59661

closed 2015-02-17 by smathermather

smathermather, 2015-02-22:
Here's the original issue: https://github.com/OpenDroneMap/OpenDroneMap/issues/33