OpenDroneMap/ODM #60: example how to use the parameters are needed

soler2000, 2015-02-21:
If for example I want to run two parameters should the parameters be used in the following format
./run.pl —odm_meshing-maxVertexCount=150000 —odm_meshing-octreeDepth=11 

or

./run.pl —odm_meshing-maxVertexCount 150000 —odm_meshing-octreeDepth 11 

what is the correct syntax for this?

smathermather, 2015-02-21:
The correct syntax is a space between the parameter and its value:
```./run.pl --odm_meshing-maxVertexCount 150000 --odm_meshing-octreeDepth 11```

soler2000, 2015-02-21:
added this to the wiki,  it needs some more info on each one of the parameters and what is the effect it has

dbaldwin, 2015-02-22:
@soler2000 @smathermather I'd like to assist with wiki documentation. Can you let me know how I go about this? I tried making a small edit, but it doesn't look like it took. Unless you have to approve it first.

soler2000, 2015-02-22:
You should just be able to save new edits to it.  I don't think there is any approval needed.

—
Sent from Mailbox

smathermather, 2015-02-23:
Indeed. Please edit away, @dbaldwin

okiave, 2015-02-25:
Regarding the todo of adding descriptons to parameters.
Here is the documentation page for PMVS v2.
http://www.di.ens.fr/pmvs/documentation.html

smathermather, 2015-02-25:
That's very helpful.

closed 2016-10-12 by dakotabenjamin