OpenDroneMap/ODM #62: Missing libvl.so?

BangorSkis, 2015-02-25:
Hi there,

Having an issue with getting ODM to run - I get the following error after the images resize:

```
1/109 - /home/ubuntu/opendronemap 2/../OpenDroneMap/bin/vlsift: error while loading shared libraries: libvl.so: cannot open shared object file: No such file or directory
parallel: Starting no more jobs. Waiting for 1 jobs to finish. This job failed:
echo -n "1/109 - " && convert -format pgm "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/G0026730_lens_geotag.jpg" "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/G0026730_lens_geotag.pgm" && "/home/ubuntu/opendronemap 2/../OpenDroneMap/bin/vlsift" "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/G0026730_lens_geotag.pgm" -o "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/G0026730_lens_geotag.key.sift" > /dev/null && perl "/home/ubuntu/opendronemap 2/../OpenDroneMap/bin/../convert_vlsift_to_lowesift.pl" "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/G0026730_lens_geotag" && gzip -f "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/G0026730_lens_geotag.key" && rm -f "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/G0026730_lens_geotag.pgm" && rm -f "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/G0026730_lens_geotag.key.sift"

quitting cause: 
        "/home/ubuntu/opendronemap 2/../OpenDroneMap/bin/parallel" --halt-on-error 1 -j+0 < "/home/ubuntu/opendronemap 2/reconstruction-with-image-size-2400/_vlsift.templist.txt"
returned with code 32512
```

I'm trying to run this in an Amazon EC2 instance so I'm not hogging my desktop machine - I managed to get it to work in a Ubuntu 12.04 VM locally, but have tried the same install on both 12.04 and 14.04 on Amazon with the same results.

Any advice?

smathermather, 2015-02-25:
Hi @BangorSkis -- don't know if that's Bangor, Maine, but I lived for a few years in Bar Harbor.

Anyway, more important things here:

What do you get when you run:
```locate libvl.so```

BangorSkis, 2015-02-25:
Exactly - Never lived in Maine (Vermont now), but wore enough plaid in high school that everyone thought I was a lumberjack from Bangor, and the nickname stuck!

Here's the output:

```
ubuntu@ip-172-31-18-121:~$ locate libvl.so
/home/ubuntu/OpenDroneMap/lib/libvl.so
/home/ubuntu/OpenDroneMap/src/vlfeat/bin/glnx86/libvl.so
/home/ubuntu/OpenDroneMap/src/vlfeat/bin/glnxa64/libvl.so
/home/ubuntu/OpenDroneMap/src/vlfeat/toolbox/mex/mexa64/libvl.so
/home/ubuntu/OpenDroneMap/src/vlfeat/toolbox/mex/mexglx/libvl.so
```

smathermather, 2015-02-25:
Weird. Ok try 

```sudo ldconfig```

and try a rerun.

(Please excuse the question) You did run ```./install.sh```, yes? I'm guessing yes, as you have the shared libraries in bin, but just to make sure you didn't copy the directory from elsewhere and run without running install.

BangorSkis, 2015-02-25:
Yes, definitely did the ./install.sh - didn't notice any errors there that I recall. Did a git clone of the top /OpenDroneMap/OpenDroneMap to get all the data.

did the sudo ldconfig, reran, same failure as a result.

BangorSkis, 2015-02-25:
Interesting, I just tried to start again from scratch and it didn't actually finish the installer - looks like it hung up @ building CMVS

```
  > getting the sources
    - getting http://ftp.gnu.org/gnu/parallel/parallel-20141022.tar.bz2
######################################################################## 100.0%
    - finished parallel.tar.bz2

    - already downloaded http://www.netlib.org/clapack/clapack-3.2.1-CMAKE.tgz
    - already downloaded http://phototour.cs.washington.edu/bundler/distr/bundler-v0.4-source.zip
    - already downloaded http://www.cs.jhu.edu/~misha/Code/PoissonRecon/Version2/PoissonRecon.zip
    - already downloaded http://www.vlfeat.org/download/vlfeat-0.9.13-bin.tar.gz
    - already downloaded http://www.di.ens.fr/cmvs/cmvs-fix2.tar.gz
    - already downloaded http://smathermather.github.io/BundlerTools/patched_files/src/graclus/graclus1.2.tar.gz
    - getting https://github.com/PointCloudLibrary/pcl/archive/pcl-1.7.2.tar.gz
######################################################################## 100.0%
    - finished pcl.tar.gz

  < done - Wed Feb 25 17:12:53 UTC 2015

  - unzipping sources
  < done - Wed Feb 25 17:13:16 UTC 2015

  - copying patches

  < done - Wed Feb 25 17:13:16 UTC 2015

  - building

  > graclus
    - cleaning graclus
    - building graclus
  < done - Wed Feb 25 17:13:28 UTC 2015

  > poisson surface reconstruction 
    - building poisson surface reconstruction
  < done - Wed Feb 25 17:13:34 UTC 2015

  > parallel
    - configuring parallel
    - building paralel
  < done - Wed Feb 25 17:13:35 UTC 2015

  > clapack
    - building clapack
    - installing clapack
  < done - Wed Feb 25 17:14:08 UTC 2015

  > vlfeat
    - installing vlfeat
  < done - Wed Feb 25 17:14:08 UTC 2015

  > cmvs/pmvs
    - cleaning cmvs
    - building cmvs
ubuntu@ip-172-31-18-121:
```

..... oh. Well the output of cmvs_2_build.log is interesting.

```
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include pmvs2.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/image/camera.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/image/image.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/image/photo.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/image/photoSetS.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/detectFeatures.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/detector.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/dog.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/expand.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/filter.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/findMatch.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/harris.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/optim.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/option.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/patch.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/patchOrganizerS.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/point.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/pmvs/seed.cc
g++ -c -O2 -Wall -Wno-deprecated -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include ../base/numeric/mylapack.cc
g++ -c -O2 -Wall -Wno-deprecated -DNUMBITS=64 -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include -I/home/ubuntu/OpenDroneMap/include/metisLib/ -fopenmp -DNUMBITS=64  cmvs.cc
g++ -c -O2 -Wall -Wno-deprecated -DNUMBITS=64 -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include -I/home/ubuntu/OpenDroneMap/include/metisLib/ -fopenmp -DNUMBITS=64  ../base/cmvs/bundle.cc
g++ -c -O2 -Wall -Wno-deprecated -DNUMBITS=64 -I/usr/local/include -I/home/ubuntu/OpenDroneMap/include -I/home/ubuntu/OpenDroneMap/include/metisLib/ -fopenmp -DNUMBITS=64  ../base/cmvs/graclus.cc
g++ -o genOption genOption.cc
../base/pmvs/optim.cc: In static member function ‘static double PMVS3::Coptim::my_f_ssd(const gsl_vector*, void*)’:
../base/pmvs/optim.cc:762:9: warning: variable ‘flag’ set but not used [-Wunused-but-set-variable]
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
make: *** [seed.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [image.o] Error 1
make: *** [dog.o] Error 1
make: *** [expand.o] Error 1
make: *** [filter.o] Error 1
make: *** [findMatch.o] Error 1
make: *** [harris.o] Error 1
make: *** [optim.o] Error 1
make: *** [patchOrganizerS.o] Error 1
make: *** [bundle.o] Error 1
```

So best guess is I ran out of RAM and didn't notice that the script didn't finish last time?

I'm using the free tier Amazon server with 1 CPU and 1GB of ram. I could fire up an instance that I'd have to pay per hour to use with more RAM, but would be ideal if there's a workaround?

smathermather, 2015-02-25:
RAM is definitely a limiting factor. I didn't realize that was the case for install (that strikes me as PMVS bug, but nonetheless...), but I know it is for running ODM on anything other than trivial datasets. I'd bump up to at least 4GB. TBH, if you are doing hourly anyway, you might bump it up to something more like 32GB, and you'll probably only have to run it for a couple hours on a given dataset. Also, that will get you SSDs which probably wouldn't hurt (though admittedly I don't know if there are any IO limited portions of the codebase).

In the mean time, if you don't mind, add 4GB as an installation pre-requisite in the [wiki](https://github.com/OpenDroneMap/OpenDroneMap/wiki), and open a separate issue on the out of memory for PMVS. I'll close this one once you've done that and link the two, and flag it as a bug.

Thanks!

BangorSkis, 2015-02-25:
Done!

I'm going to try and enable a swap file first and see if I can build in enough SSD-based virtual RAM to stay within the free server - if it doesn't work I'll pay the something like $0.20/hr to run a much more powerful VM, haha!

Thanks! if I can get this working on the EC2 cloud it will be awesome and hopefully SUPER fast.

smathermather, 2015-02-25:
Cool. Closed, referencing: https://github.com/OpenDroneMap/OpenDroneMap/issues/63

closed 2015-02-25 by smathermather