OpenDroneMap/ODM #67: Error code 32512

okiave, 2015-03-02:
I tested ODM on two different machines running ubuntu 64 trusty
and got the same errors. After the ODM finishes resizing all the images
and moves on to its next step it brings up the following error in terminal:

> - finding keypoints - Mon Mar 2 22:17:59 UTC 2015

> sh: 1: /home/vagrant/klatt/../OpenDroneMap/bin/parallel: not found

> quitting cause:
	"/home/vagrant/klatt/../OpenDroneMap/bin/parallel" --halt-on-error 1 -j+0 < "/home/vagrant/klatt/reconstruction-with-image-size-2400/_vlsift.templist.txt"
returned with code 32512

***Contents of _vlsift.templist.txt (last line)***
convert_vlsift_to_lowesift.pl" "/vagrant/klatt/reconstruction-with-image-size-2400/3882_klatt-park" && gzip -f "/vagrant/klatt/reconstruction-with-image-size-2400/3882_klatt-park.key" && rm -f "/vagrant/klatt/reconstruction-with-image-size-2400/3882_klatt-park.pgm" && rm -f "/vagrant/klatt/reconstruction-with-image-size-2400/3882_klatt-park.key.sift"

I checked the bin for parallel directory and it exists.
Any ideas? thx

smathermather, 2015-03-03:
How much RAM are you working with?

okiave, 2015-03-03:
One machine is running 5gb the other 20gb. There's also plenty of HD space
on both machines.

smathermather, 2015-03-03:
Very odd. Did you try running:

```sudo ldconfig```?

okiave, 2015-03-03:
I ran another test on both machines where I increased the CPU numbers. The workstation got further than the laptop, but gave the same 32512 error after being 10 images into the keypoint matching.

>1/44 - /bin/sh: 1: /vagrant/klatt/../OpenDroneMap/bin/vlsift: not found
>parallel: Starting no more jobs. Waiting for 12 jobs to finish. This job failed

I tried sudo ldconfig and that changed. no dice. 
It seems the error is happening when parallel starts doing its thing. 
Is there a place online I can go to cross reference the 32512 code?
I tried google but I don't exactly understand what Parallel is.

okiave, 2015-03-03:
Upon re-reading this <a href="https://github.com/OpenDroneMap/OpenDroneMap/issues/62">thread</a>, i'm going to try a fresh reinstall and see how it goes.

okiave, 2015-03-04:
**Conclusion**
When using vagrant with virtualbox to create guest machine, note that
virtualbox defaults to 512mb of RAM for the guest machine. This 
limited RAM created the issue that led to the error.

**Solution**
After getting the guest box up using vagrant, shut it down using *vagrant halt*, 
then up the RAM accordingly to your system. After that you *vagrant up* with the
new settings and run the ./install.sh for OpendDroneMap.

You can read more about the error at the thread my comment above.

smathermather, 2015-03-04:
Ah, makes sense. It won't even install without a fair amount of RAM.

closed 2015-03-04 by smathermather

jewelthief, 2015-07-24:
Well I am seeing 32512 as well, here is current state of memory in the remote VM box. Please advice.

MemTotal:        1716656 kB
MemFree:         1035708 kB
MemAvailable:    1322184 kB
Buffers:           32232 kB
Cached:           418316 kB
SwapCached:            0 kB
Active:           222000 kB
Inactive:         348936 kB
Active(anon):     121100 kB
Inactive(anon):     8440 kB
Active(file):     100900 kB
Inactive(file):   340496 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:        120424 kB
Mapped:            78408 kB
Shmem:              9160 kB
Slab:              47328 kB
SReclaimable:      28032 kB
SUnreclaim:        19296 kB
KernelStack:        4224 kB
PageTables:         9616 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      858328 kB
Committed_AS:     950024 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       53696 kB
VmallocChunk:   34359682548 kB
HardwareCorrupted:     0 kB
AnonHugePages:     40960 kB
CmaTotal:              0 kB
CmaFree:               0 kB
HugePages_Total:       0
HugePages_Free:        0
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
DirectMap4k:       77760 kB
DirectMap2M:     1757184 kB