OpenDroneMap/ODM #68: Which dynamic libraries should be used for bundler?

radarku, 2015-03-03:
On a fresh Ubuntu 12.04.5 LTS VM, I took two paths to using bundler unsuccessfully.

1) using the packages from the apt-repository results in a Seg Fault in bundler
```
...
  Adding match 72100 ==> 71069 [15001]
 tri.error[15008] = 1.894
[SifterApp::DumpOutputFile] Wrote file in 0.090s
initial motstr-SBA error 5799.19 [0.193281]

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff6d7e018 in dscal_ () from /usr/lib/libblas.so.3gf
(gdb) bt
#0  0x00007ffff6d7e018 in dscal_ () from /usr/lib/libblas.so.3gf
#1  0x00007ffff73cc1a3 in dsytf2_ () from /usr/lib/liblapack.so.3gf
#2  0x00007ffff73cebf4 in dsytrf_ () from /usr/lib/liblapack.so.3gf
#3  0x00000000004b0620 in sba_symat_invert_BK ()
#4  0x00000000004a8362 in sba_motstr_levmar_x ()
#5  0x00000000004a3d89 in sba_motstr_levmar ()
#6  0x00000000004a1db5 in run_sfm ()
#7  0x00000000004187fe in BundlerApp::RunSFM(int, int, int, bool, camera_params_t*, v3_t*, int*, v3_t*, std::vector<std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > >, std::allocator<std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > > >&, double, double*, double*, double*, double*, bool) ()
#8  0x000000000042bfd4 in BundlerApp::BundleAdjustFast() ()
#9  0x000000000040616e in BundlerApp::OnInit() ()
#10 0x0000000000403a7e in main ()
```
2) using the libraries bundled with OpenDroneMap, bundler is unable to link libgfortran.so.1

No packages in the Ubuntu repositories, AFAIK, have lower than libgfortran.so.3

```
OpenDroneMap/bundler-v0.4-source$ LD_LIBRARY_PATH="OpenDroneMap/lib/" ldd bin/bundler
        linux-vdso.so.1 =>  (0x00007fffb8f24000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f5160df1000)
        liblapack.so.3gf => /opt/OpenDroneMap/lib/liblapack.so.3gf (0x00007f5160704000)
        libblas.so.3gf => /opt/OpenDroneMap/lib/libblas.so.3gf (0x00007f51604b2000)
        libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f5160262000)
        libANN_char.so => /opt/OpenDroneMap/lib/libANN_char.so (0x00007f516004f000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f515fd4e000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f515fa52000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f515f83c000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f515f47c000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f5161019000)
        libgfortran.so.1 => not found
        libgfortran.so.1 => not found
```

It might be easiest to just include libgfortran.so.1 with the libraries included in OpenDroneMap/lib/ but I can't seem to dig it up.

smathermather, 2015-03-03:
If you start with a fresh machine, you should be able to just run:
```./install.sh``` from the clone of the repo. This will do all the installation for you.

radarku, 2015-03-03:
Thanks, and sorry for not explaining the whole story.  The install goes fine.  When I ran the 'run.pl' script against some actual data, I got to the step that runs bundler, and got a SegFault.  The snippets are the results of my debugging as to why it SegFaults.

smathermather, 2015-03-04:
When you run the install, what does it say at the end?

smathermather, 2015-03-04:
Also, this might be related: https://github.com/OpenDroneMap/OpenDroneMap/issues/67

radarku, 2015-03-04:
It looks like the install wasn't actually completing.  pcl has an error building:
```
Scanning dependencies of target pcl_common
[  1%] Building CXX object common/CMakeFiles/pcl_common.dir/src/point_types.cpp.o
/opt/OpenDroneMap/src/pcl/common/src/point_types.cpp:1:0: error: CPU you selected does not support x86-64 instruction set
make[2]: *** [common/CMakeFiles/pcl_common.dir/src/point_types.cpp.o] Error 1
make[1]: *** [common/CMakeFiles/pcl_common.dir/all] Error 2
make: *** [all] Error 2
```
Could that be related?

smathermather, 2015-03-04:
Odd. It should work on 32-bit architecture, but perhaps we have a regression I am not aware of. Are you running with 32-bit Ubuntu?

smathermather, 2015-03-06:
That definitely sounds related. I'm confused by the error though -- in my
tests, it compiles fine on 32-bit machines. Have you tried with a 64-bit VM?

radarku, 2015-03-06:
The errors I was having were on a 64-bit VM.  I started over with a 32-bit VM and had much better luck.  The output looks a bit strange, though.  Likely because either (1) I guessed the wrong focal length for my camera, or (2) I neglected to correct for the lens distortion of the GoPro.

smathermather, 2015-03-06:
GoPro definitely needs lens correction.

radarku, 2015-03-07:
After lens distortion correction, the output is quite nice.  For others, I found the imagemagick tools to be extremely helpful:  http://www.imagemagick.org/Usage/lens/

```
   convert distorted.jpg -distort barrel 'a b c' corrected.jpg
```
...where a, b, and c are the distortion calibration factors.

2 things that I needed to do:
(1) all the source files must have a lowercase .jpg extension
(2) GNU parallel was consistently failing at one step, so I needed to run it serially:
```
<  run("\"$BIN_PATH/parallel\" --halt-on-error 1 -j+0 < \"$jobOptions{step_1_vlsift}\"");
>  run("/bin/sh \"$jobOptions{step_1_vlsift}\"");
```

nirabpudasaini, 2015-03-08:
It seems like more core you have more RAM you need. Maybe there sould be an option to limit the number of parallel jobs for the KeyPoint generation stage. I was also having the issue with parallel in wich two jobs were failing and two were completing sucessfully.

Preston-HackSpace, 2015-04-05:
I also got a Segmentation Fault (core dumped) messaged when the script started run bundler. I installed Ubuntu 12 64 bit on Oracle Virtual Machine. The virtual machine has 8GB RAM and 5 out of 8 cores assigned on a i7 processor.

Installation was all fine, no error messages. I followed all the instructions as suggested.

Is there an issue with using a 64bit operating system?

smathermather, 2015-04-22:
Sorry for the delay @Preston-HackSpace. It should run fine on Ubuntu 12 64-bit, but you the issue could be related to what @nirabpudasaini mentions above (ratio of RAM to processors).

Can you try allocating 2 processors and see how it runs, or double your RAM? We need to get metrics / rules of thumb for how much RAM per processor for bundler, but to my knowledge, no one has run those tests yet.

smathermather, 2015-11-23:
Let me know if this needs re-opened.

closed 2015-11-23 by smathermather