OpenDroneMap/ODM #69: Focal length for DJI Inspire 1 Camera

hetlelid, 2015-03-03:
Thanx for an awesome toolkit :)

I'm unsure about what to add to the ccd_defs.json file. Can anyone confirm that 3.61mm is the correct value? Sounds low compared to other cameras. Should i use 20mm (35mm format equivalent)?

CMOS 1/2.3”
Specs: http://www.dji.com/product/inspire-1/spec.

hetlelid, 2015-03-03:
It seems like i have to go for CCD 6.17, is this correct?

okiave, 2015-03-04:
@hetlelid go with the specs on the DJI page and convert
the number to metric. Don't worry about focal length. 
When you modify the <a href="https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/ccd_defs.json">ccd_defs.json</a> write it like the GoPro entry, but w/ your
cam specs.

closed 2015-11-23 by smathermather