OpenDroneMap/ODM #74: Error in OdmTexturing

KhanTengri1, 2015-03-06:
Hi All,
I received an ... Error in OdmTexturing...
had tried the odm_boruszyn_kap dataset...

tail from odm_texturing_log.txt:
/build/buildd/opencv-2.4.8+dfsg1/modules/core/src/matrix.cpp:323: error: (-215) 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows in function Mat

had no errors when I tried the seneca dataset a week ago...

I've got Ubuntu 14.04 64, install finished ok... with the version of opendronemap I cloned on 28.02

Has anyone an idea what i did wrong?

Many thx
Robert Wagner

merkato, 2015-03-09:
Hi,
Originally, I have same error with this dataset, before it was uploaded to github. 

But right now, on another instalation of Ubuntu 14.04 64, with 8GB RAM, this dataset was computed in only 30 minutes, with no problem. Maybe that's memory problem, as Stephen was suggesting....

KhanTengri1, 2015-03-10:
I'll to install some more ram...
thx
Robert

closed 2015-03-10 by KhanTengri1

SimFaris, 2015-10-15:
@merkato I got the same error, maybe you can rotate the pic img_4909.jpg. when I do this, I found I got the correct result. In a word, I thought the ODM got the wrong size when it calculates the texture, but opencv assert that.

smathermather, 2015-10-15:
Very interesting @SimFaris.