OpenDroneMap/ODM #76: UtmExtractor fails to read the EXIF data

nirabpudasaini, 2015-03-09:
I see that there is an option --odm_georeferencing-useGcp that can be set to false to skip reading from the GCP file.

/OpenDroneMap/run.pl --odm_georeferencing-useGcp "false" this command sould skip reading GCP and use the EXIF to generate the cordList.txt for georefrencing.

However the UtmExtractor is not working correctly for me. It fails to find the Latitude and throws exception despite my images having all the data. This is the output when i rung jhead on the image file.

File name : DJI02089.jpg
File size : 11324087 bytes
File date : 2015:03:08 10:58:31
Camera make : DJI
Camera model : PHANTOM VISION FC200
Date/Time : 2015:02:05 10:51:17
Resolution : 4608 x 3456
Focal length : 5.0mm
Exposure time: 0.0006 s (1/1667)
Aperture : f/2.8
ISO equiv. : 100
Whitebalance : Auto
Metering Mode: center weight
Exposure : program (auto)
Jpeg process : Progressive
GPS Latitude : N 27d 43m 33.94s
GPS Longitude: E 85d 20m 33.11s
GPS Altitude : 1352.30m
JPEG Quality : 100

You can see that there is Latitude Longitude and Altitude but still the UtmExtractor exist with log entry

Error in OdmExtractUtm:
Image is missing GPS Latitude data

My data can be found at https://github.com/nirabpudasaini/mission01/tree/master/LensCorrection

seanXYZ, 2015-09-11:
Hi, 
I think you misunderstand the meaning of GCP. GCP means Ground Control Point. So, the Latitude Longitude and Altitude in your JPG file is not a GCP file input. In your case, you should try -coordFile to input the cameras GPS  georeference！

smathermather, 2015-09-11:
No, @nirabpudasaini is correct in assuming the logic that --odm_georeferencing-useGcp set to false should be forcing the use of EXIF data from the jpeg.

Which branch are you using, @nirabpudasaini?

seanXYZ, 2015-09-11:
Oh, my god! Sorry for my fault missing " set to false "

seanXYZ, 2015-09-11:
how to write camera parameters and GPS into a JPG file:  what format EXIF should be like?  Because I use a  camera with no name(I even do not know where I get it).
1>  I have already calibrate it by using OpenCV, so  I get camera model like [ fx  0  cx; 0  fy cy; 0  0  1] and distortion parameters [d0  d1 d2 d3 d4]. 
2>  I have got GPS Longitude Latitude Attitude from my UAV.
3>  no camera sense  parameters like "Focal length : 5.0mm"
How can I input these information in to JPG file ？ and make sure this information can be parsed and used by ODM?

smathermather, 2015-09-11:
Hi @seanXYZ --

Pre calibration is great, and probably reduces noise in the overall result. ODM will attempt to calibrate it as well, but you'll already have the images converted to a pinhole model, which is great.

ODM doesn't need to know those parameters, nor does it need to know your focal length -- it will calculate that. It does need to know the size of your CMOS or CCD chip. You can theoretically specify this with the ```--force-ccd`` flag. That flag was broken until recently, but I think it was fixed (hailing @dakotabenjamin).

dakotabenjamin, 2015-09-11:
If you have a .gpx file from the UAV, you can use Mapillary's Python tools
to geotag your photos. Here are some links to help:

http://blog.amanzi.org/2014/05/using-gopro-time-lapse-photos-for.html?m=1

https://github.com/mapillary/mapillary_tools/tree/master/python

Other ways: http://wiki.openstreetmap.org/wiki/Geotagging_Source_Photos

smathermather, 2015-09-11:
If the ```--force-ccd``` flag _is_ still broken, you could create a dummy sensor name in the images using an exif command line tool of your choice, and then add that dummy sensor name to the [JSON file containing the list of cameras](https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/ccd_defs.json). It's a cludge, but an easy one. If you need more details / walk-through, let me know.

But... try ```--force-ccd``` first.

dakotabenjamin, 2015-09-11:
As far as I remember --force-ccd works.

seanXYZ, 2015-09-14:
Thank you a lot~

seanXYZ, 2015-09-15:
@smathermather  It works when using --force-ccd:
 ../OpenDroneMap-gh-pages/run.pl --odm_georeferencing-useGcp false --start-with bundler --force-ccd 6.20

smathermather, 2015-09-17:
Hey! Awesome news.

closed 2015-09-17 by smathermather