OpenDroneMap/ODM #77: Segmentation fault (core dumped)

cgdnorth, 2015-03-10:
Hello, I am getting the error ```Segmentation fault (core dumped)``` in running 4 images as a test. Can you give me a reason this is happening? The setup is Ubuntu 12.04 running in VirtualBox on a Mac.
```
Segmentation fault (core dumped)

quitting cause: 
	"/home/chris/process/corrected/../../OpenDroneMap/bin/bundler" "/home/chris/process/corrected/reconstruction-with-image-size-2400/list.txt" --options_file "/home/chris/process/corrected/reconstruction-with-image-size-2400/options.txt" > bundle/out
returned with code 35584
```

smathermather, 2015-03-16:
Hi @cgdnorth -- can you post this to the user forum?

Thanks!
http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users

closed 2015-03-16 by smathermather

gutzmann, 2015-09-30:
what was ever the resolution on this?

dakotabenjamin, 2015-10-01:
Yes, check #171