OpenDroneMap/ODM #79: Messing with PATH...

FLYBYME, 2015-03-11:
Trying to install and not impressed...

```
root@6e4ca79fb3e1:/# rm
rm: missing operand
Try `rm --help' for more information.
root@6e4ca79fb3e1:/# git clone https://github.com/OpenDroneMap/OpenDroneMap.git /OpenDroneMap
Cloning into '/OpenDroneMap'...
remote: Counting objects: 1047, done.
remote: Total 1047 (delta 0), reused 0 (delta 0), pack-reused 1047
Receiving objects: 100% (1047/1047), 480.63 MiB | 22.79 MiB/s, done.
Resolving deltas: 100% (403/403), done.
root@6e4ca79fb3e1:/# rm
rm: missing operand
Try `rm --help' for more information.
root@6e4ca79fb3e1:/# rm
rm: missing operand
Try `rm --help' for more information.
root@6e4ca79fb3e1:/# /OpenDroneMap/install.sh

     created by Daniel Schwarz/daniel.schwarz@topoi.org
     released under Creative Commons/CC-BY
     Attribution

     if the script doesn't finish properly
     (i.e. it doesn't print "script finished" at the end)
     please email me the content of the logs folder

  - script started - Wed Mar 11 19:22:19 UTC 2015
sudo: rm: command not found
root@6e4ca79fb3e1:/# rm
bash: /bin/rm: No such file or directory
root@6e4ca79fb3e1:/#

```

dbaldwin, 2015-03-11:
What distribution are you running? Also, see the note about the contents of the log folder. Is there any output in there?

FLYBYME, 2015-03-11:
ubuntu:12.04

```
root@9d4d49ac642e:/# cd /OpenDroneMap/
root@9d4d49ac642e:/OpenDroneMap# ls
bash: ls: command not found
root@9d4d49ac642e:/OpenDroneMap# cd^C
root@9d4d49ac642e:/OpenDroneMap# cd   
.git/                          cmvs.tar.gz.part-ac            odm_extract_utm/
.gitignore                     cmvs.tar.gz.part-ad            odm_georef/
PoissonRecon.zip               cmvs.tar.gz.part-ae            odm_meshing/
README.md                      code_of_conduct.md             odm_orthophoto/
bundler.zip                    convert_vlsift_to_lowesift.pl  odm_texturing/
ccd_defs.json                  graclus.tar.gz                 patched_files/
clapack.tgz                    img/                           run.pl
cmvs.tar.gz.part-aa            install.sh                     vlfeat.tar.gz
cmvs.tar.gz.part-ab            licenses/                      
root@9d4d49ac642e:/OpenDroneMap# cd 
.git/                          cmvs.tar.gz.part-ac            odm_extract_utm/
.gitignore                     cmvs.tar.gz.part-ad            odm_georef/
PoissonRecon.zip               cmvs.tar.gz.part-ae            odm_meshing/
README.md                      code_of_conduct.md             odm_orthophoto/
bundler.zip                    convert_vlsift_to_lowesift.pl  odm_texturing/
ccd_defs.json                  graclus.tar.gz                 patched_files/
clapack.tgz                    img/                           run.pl
cmvs.tar.gz.part-aa            install.sh                     vlfeat.tar.gz
cmvs.tar.gz.part-ab            licenses/                      
root@9d4d49ac642e:/OpenDroneMap# cd logs 
bash: cd: logs: No such file or directory
root@9d4d49ac642e:/OpenDroneMap# ls
bash: ls: command not found
root@9d4d49ac642e:/OpenDroneMap# ls
bash: ls: command not found
root@9d4d49ac642e:/OpenDroneMap# cd ~/
root@9d4d49ac642e:~# cd .
./        ../       .bashrc   .profile  

```

Im pissed by the way ls rm what else is missing...

FLYBYME, 2015-03-11:
Do you see a log file??

dbaldwin, 2015-03-11:
Yeah that's crazy. Is something up with your path because I'm assuming it's failing early in the script like here where it's using the "rm" command:

https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/install.sh#L78

If you can't "ls" or "rm" see if this helps:

http://askubuntu.com/questions/186808/every-command-fails-with-command-not-found-after-changing-bash-profile

FLYBYME, 2015-03-11:
Problem was:
`/OpenDroneMap/install.sh`

When I should have:
`cd /OpenDroneMap;./install.sh`

wolkstein, 2015-03-12:
hi,
@FLYBYME 
why you are root? i do not know the way you access root. but maybe this is the problem because your environment variables as root(in this case) are not set. try to do all install steps as normal user.

nirabpudasaini, 2015-03-13:
Most probably there is some problem with the $PATH variable in your Ubuntu. Or you have managed to mess up your /bin folder for the root user. You should look for help in the Ubuntu forums and make sure that basic commands like rm , ls are working properly. 

Please do a "echo $PATH" and post what you get back.

FLYBYME, 2015-03-14:
I was trying to install it in a docker container.

smathermather, 2015-03-16:
Interesting. I don't know anything about the ins and outs of docker to recommend settings.

smathermather, 2015-03-16:
@Flybyme -- can you close this and open an issue requesting an OSM Docker container? I think that would be a very useful issue.

smathermather, 2015-11-23:
@FLYBYME -- let me know if this needs re-opened.

closed 2015-11-23 by smathermather