OpenDroneMap/ODM #81: Blending options

wildintellect, 2015-03-15:
Image seam blending seems fine but misses color and exposure blending. Hugin uses http://enblend.sourceforge.net/ but perhaps there's also something more native to the OpenCV pipeline being discussed.

smathermather, 2015-03-16:
Whoa! This shows some real potential.

I wonder if this would be a step that only runs at the orthophoto step, since at that stage the rendered images have no parallax issues.

smathermather, 2015-03-16:
Cool. Looks like it's GPL too, which is compatible, should we ever build a binary.

Regarding OpenCV -- no such discussion.

smathermather, 2015-11-23:
We'll use OpenCV for blending. Spotscale is currently working on this, so I'm going to close this Enblend issue.

closed 2015-11-23 by smathermather