OpenDroneMap/ODM #85: OpenCV error when running odm_texturing

naktinis, 2015-03-25:
I'm running OpenDroneMap (simply run.pl without any parameters) on the "boruszyn_kap" sample data (https://github.com/OpenDroneMap/odm_data) and I get the following error:

```
<...>

  - running meshing - Wed Mar 25 15:59:58 GMT 2015

  - running texturing - Wed Mar 25 16:00:32 GMT 2015

OpenCV Error: Assertion failed (0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows) in Mat, file /foo/code/opencv/modules/core/src/matrix.cpp, line 323
Error in OdmTexturing:
/foo/code/opencv/modules/core/src/matrix.cpp:323: error: (-215) 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows in function Mat

For more detailed information, see log file.

quitting cause: 
	"/foo/code/OpenDroneMap/bin/odm_texturing" -bundleFile /foo/programs/odm_data/boruszyn_kap/reconstruction-with-image-size-2400/pmvs/bundle.rd.out -imagesPath /foo/programs/odm_data/boruszyn_kap/ -imagesListPath /foo/programs/odm_data/boruszyn_kap/reconstruction-with-image-size-2400/pmvs/list.rd.txt -inputModelPath /foo/programs/odm_data/boruszyn_kap/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -outputFolder /foo/programs/odm_data/boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/ -textureResolution 4096 -bundleResizedTo 2400 -textureWithSize 3600 -logFile /foo/programs/odm_data/boruszyn_kap/reconstruction-with-image-size-2400/odm_texturing/odm_texturing_log.txt
returned with code 256
```

dbaldwin, 2015-03-25:
What are the specs (memory, OS version) on your setup? See this as it may help:

https://github.com/OpenDroneMap/OpenDroneMap/issues/74

Fi156, 2015-04-16:
Got the same error, but can provide some more infos:

CPU: AMD FX-8320
RAM: approx. 10GB

Linux Mint 17.1 (based on Ubuntu 14.04) 64-bit
Kernel: 3.16.2

OpenCV Version: 2.4.8

Full log from run.pl:
http://pastebin.com/X39ji9ES

Full log from "Texturing":
http://pastebin.com/ehSD11MH

Memory and CPU usage
(created with "sar -ru"):
http://pastebin.com/uhUrdjhA

TL;DR:
 In the texturing step Memory occupation does not exceed 65%.

burakaydn, 2015-05-21:
I got the same error, any solutions?

smathermather, 2015-05-21:
Can you post your mesh and ply file?

burakaydn, 2015-05-21:
Log Files:
odm_texturing_log.txt
http://ge.tt/7CTxfqG2/v/0
odm_meshing_log.txt 
http://ge.tt/2p9NhqG2/v/0

ply file under results folder
http://ge.tt/4tHQhqG2/v/0

I can't figure out with which file you mentioned as mesh, sorry.

smathermather, 2015-05-22:
Ok, a couple things to try, but let's start with this one:

1. Bring that ply into MeshLab and remove non-manifold edges. Save back out as a ply, and continue running ODM from the texturing step with ```run.pl --start-with odm_texturing```.

burakaydn, 2015-05-22:
Same error :(

smathermather, 2015-05-25:
Weird. Try rerunning with resize set to 1200.

smathermather, 2015-05-26:
Actually, @burakaydn I should check -- are images named with a .JPG extension or lower case .jpg?

burakaydn, 2015-05-26:
lower case actually i showed Meshlab screen as proof of concept and its enough for me, thank you so much.

smathermather, 2015-05-27:
I still wish I knew why it was causing you problems. I'll close the ticket, but may dive into this more when I get the chance.

SimFaris, 2015-08-15:
@burakaydn  @smathermather I have got the same error, Could you give me some advice. Thank U.

Fi156, 2015-08-28:
Just a quick update:

A fresh version of the python branch was pulled before.

Dataset: boruszyn_kap
Started with --resize 1200 and Meshlab did not find any non-manifold edges.

Other datasets are working fine.

SimFaris, 2015-10-15:
maybe you can rotate the pic img_4909.jpg. good luck.

closed 2016-10-12 by dakotabenjamin