OpenDroneMap/ODM #91: running meshing: Segmentation fault (core dumped)

joseluisbossio, 2015-04-05:
Hi, we are trying OpenDroneMap over Linux Ubuntu 14.04 x64 with the examples called 'caliterra_101914_hexa', that was published in https://www.dropbox.com/sh/1rt1qdh7tw2mybi/AAAv7zukDC7fszyJMXfc5x8wa?dl=0.

Unfortunately we have a core dump during the phase 'running meshing':
...
  - running meshing - dom abr 5 11:00:09 ART 2015
Segmentation fault (core dumped)

quitting cause: 
	"/home/administrator/odm_app/OpenDroneMap/bin/odm_meshing" -inputFile /home/administrator/Downloads/odm_app/caliterra/reconstruction-with-image-size-2400-results/option-0000.ply -outputFile /home/administrator/Downloads/odm_app/caliterra/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -logFile /home/administrator/Downloads/odm_app/caliterra/reconstruction-with-image-size-2400/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1
returned with code 35584

...
more /home/administrator/Downloads/odm_app/caliterra/reconstruction-with-image-size-2400/odm_meshing/odm_meshing_log.txt

Reading point cloud at: /home/administrator/Downloads/odm_app/caliterra/reconstruction-with-image-size-2400-results/option-0000.ply
Writing output to: /home/administrator/Downloads/odm_app/caliterra/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply
Writing log information to: /home/administrator/Downloads/odm_app/caliterra/reconstruction-with-image-size-2400/odm_meshing/odm_meshing_log.txt
Vertex count was manually set to: 100000
Octree depth was manually set to: 9
The number of samples per octree node was manually set to: 1
Successfully loaded 369260 points with corresponding normals from file.
Octree depth used for reconstruction is: 9
Estimated initial vertex count: 262144

Reconstruction complete:
Vertex count: 193570
Triangle count: 386952

Decimating mesh, removing 48.3391 percent of vertices.
Decimation complete.
Decimated vertex count: 100041
Decimated triangle count: 199902

Saving mesh to file.
Successfully wrote mesh to:
/home/administrator/Downloads/odm_app/caliterra/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply

Thanks for your support and Happy Easter!
Jose Luis Bossio
@jlbsensedrone

dbaldwin, 2015-04-06:
How much memory does your system have? I saw this happen when I first got started with ODM on systems with less than 8 GB RAM. FWIW I recently did a howto using this imagery and ODM on Ubuntu 14.04:

https://www.youtube.com/watch?v=82QHOihgm3U

joseluisbossio, 2015-04-06:
Thanks a lot Dennis for your quickly answer.

Unfortunately in this oportunity, our RAM memory it´s less than you propouse.

We saw the video and was very interesting and clear to understand easy how to start in OpenDroneMap.

Please, could you view our next post https://github.com/OpenDroneMap/OpenDroneMap/issues/92 about the use of GPU to process the images in ODM? 

Thanks for you time (and of course your patience)

Jose Luis Bossio from @sensedrone
@jlbsensedrone

smathermather, 2015-11-23:
Will close. Let me know if this is not resolved (if RAM was not the issue) and I'll reopen.

closed 2015-11-23 by smathermather