OpenDroneMap/ODM #92: Image Processing using GPU

joseluisbossio, 2015-04-06:
Hi, we are testing OpenDroneMap and we have decided to create a cluster to images processing. So, the question is, Can you recommend some graphics cards we could use to send the process from CPU to GPU to accelerate the process time? (maybe a nVIDIA G-Force Model)

Ofcourse, the more important question is, Do you know if OpenDroneMap was developed to support a GPU processing?

Thank you very much for share your experiences with us.

Jose Lusi Bossio from @sensedrone
@jlbsensedrone

joseluisbossio, 2015-04-06:
[image]
[image]

smathermather, 2015-04-07:
You might look to SiftGPU: http://www.cs.unc.edu/~ccwu/siftgpu/ . This is something that would be a welcome addition. I think this is NVidia dependent.

smathermather, 2015-04-07:
Looking closer, SiftGPU is licensed for (maybe?) academic / non-commercial use only. It does (optionally) do CUDA but looks like it will work with other cards.

KhanTengri1, 2015-04-16:
as far as I know SiftGPU uses the Lowe algorithm which is patented, 
further more Mr. Changchang Wu works for Google now ....
and they state at http://cs.unc.edu/~ccwu/siftgpu/ "....For commercial licensing, please contact Peter Liao (liaopb@unc.edu, 919-966-3929)...."

smathermather, 2015-04-16:
You are right, @KhanTengri1 -- I jumped down that rabbit hole last week. The correct contact for within the USA (and any other nation that recognizes software patents) is Roger.miller@uilo.ubc.ca. So SiftGPU is (I think / IANAL) definitely not compatible with GPL use.

liaopb@unc.edu is the correct contact for any licensing outside the USA.

gateway, 2016-11-09:
I'm looking at using this project but I was wondering has their been any movement on GPU support.  The idea would be to fire up a few amazon gpu instances and que up data sets.. thoughts?

taiwuchiang, 2016-12-01:
I have the same request. If someone can point me how to do it, I am willing to spend time to make it work. I got a machine with GTX 1080 there and like to use it for this project. Thanks.

smathermather, 2016-12-05:
No movement yet on GPU support. The most likely candidate portion of the software to improve here would be OpenSfM. I would open a ticket there (if there's not one already) to see what timeline if any they have for GPU support.

taiwuchiang, 2016-12-06:
If I build OpenCV with WITH_CUDA on, will it help speed up the execution?

gateway, 2016-12-06:
@taiwuchiang I think he suggested we open up a discussion with OpenSfM to see what they have available as far as GPU support.

kikislater, 2019-05-28:
Just to update this thread : things are moving forward I think : 
- Modern GPU implementation : https://github.com/Celebrandil/CudaSift => Speed process increase by x 2
- Full GPU process with OpenMVG + OpenMVS => https://www.researchgate.net/publication/328073102_Reconstructing_3D_Virtual_Environments_within_a_Collaborative_e-Infrastructure

edgarriba, 2019-05-28:
alternatives using modern DNN libraries are coming out: https://github.com/ducha-aiki/pytorch-sift
which will be soon integrated to: https://github.com/arraiyopensource/kornia
/cc @ducha-aiki

seangrant82, 2020-03-09:
since this (https://github.com/OpenDroneMap/ODM/pull/1069) was merged, I would assume this issue is now a step closer?

smathermather, 2020-03-16:
This is part of the puzzle for sure. :)

gotjoshua, 2020-04-21:
Greetings, What is the current game plan for GPU support? Will it only work for native installs or will some sort of [GPU for docker](https://marmelab.com/blog/2018/03/21/using-nvidia-gpu-within-docker-container.html) be possible?

pierotofy, 2020-04-21:
GPU for docker will probably be used (since we rely on docker to run things).

pierotofy, 2020-06-03:
We're launching a fund to get this done: https://fund.webodm.org/#/fund/3

Pledges are welcome if you'd like to get this implemented!

Saijin-Naib, 2020-06-03:
With WSL2, GPU acceleration should be possible in short order on Windows hosts with WSL2 clients!

kikislater, 2020-06-06:
WSL2 :-/ ...

pierotofy, 2021-04-05:
ODM now has GPU accelerated feature extraction capabilities. (See README for more information).

closed 2021-04-05 by pierotofy