OpenDroneMap/ODM #93: Mobius Cam

FLYBYME, 2015-04-13:
Do you think we can add support for the Mobius Cam?

smathermather, 2015-04-14:
Might be tricky. Any thoughts @wildintellect?

wildintellect, 2015-04-14:
I don't see any particular reason why not, just need a camera model for it. Cole was testing checkerboard calibration code, maybe give that a try. Not sure if it's posted yet.

FLYBYME, 2015-04-14:
I have some information here that might be of use.

https://github.com/DroidPlanner/Tower/issues/1290

smathermather, 2015-04-14:
Cool. Can you run ```jhead``` on an image from the Mobius Cam?

FLYBYME, 2015-04-14:
```
Approximate quality factor for qtable 0: 90 (scale 19.88, var 1.14)
Approximate quality factor for qtable 1: 90 (scale 20.45, var -6.08)
JPEG image is 2304w * 1536h, 3 color components, 8 bits per sample
File name    : .\IMAG1102.JPG
File size    : 1334184 bytes
File date    : 2015:03:31 17:45:40
Resolution   : 2304 x 1536
```

All I got from jhead.

smathermather, 2015-04-14:
Re: Cole, he posted code, but I don't think it's yet a full utility yet.

smathermather, 2015-04-14:
@FLYBYME -- install exif
```sudo apt-get install exif```

and run EXIF on it. It should state the name of the camera in the header too.

smathermather, 2015-04-22:
Hey @FLYBYME -- re-reading, you did provide everything needed, so I added it:

https://github.com/OpenDroneMap/OpenDroneMap/commit/c96b49c480defcc87d9622b772267acbd514794f

Cheers

closed 2015-04-22 by smathermather