OpenDroneMap/ODM #96: libdc1394 error: Failed to initialize libdc1394

chriseyhorn, 2015-04-15:
I was getting the Failed to initialize libdc1394 on the last few scripts.  This is running on an EC2 instance.  

- running meshing - Wed Apr 15 16:42:29 UTC 2015

  - running texturing - Wed Apr 15 16:42:50 UTC 2015

libdc1394 error: Failed to initialize libdc1394

  - running georeferencing - Wed Apr 15 16:46:24 UTC 2015

libdc1394 error: Failed to initialize libdc1394

  - running orthophoto generation - Wed Apr 15 16:46:30 UTC 2015

libdc1394 error: Failed to initialize libdc1394

  - done - Wed Apr 15 16:46:36 UTC 2015

smathermather, 2015-04-15:
Are your running this in a VM? If memory serves, I think this error is harmless but goes away if you enable USB on your VM.

chriseyhorn, 2015-04-15:
I'm running on a EC2 instance.

smathermather, 2015-04-15:
Does it produce all the products (point cloud, mesh, etc)?

dbaldwin, 2015-04-16:
I've seen this error many times before and it hasn't caused any issues. Just ignore it.

Here is some background:

http://stackoverflow.com/questions/12689304/ctypes-error-libdc1394-error-failed-to-initialize-libdc1394

chriseyhorn, 2015-04-16:
I will have to take a look on the instance for the data files.  I was just concerned about the error since there were no additional messages between each of the "running" messages.

closed 2015-04-16 by smathermather