OpenDroneMap/ODM #97: Error in OdmTexturing: Failed to read image

RB-RSA, 2015-04-16:
Hi There,

I get into trouble during the Odmtexturing phase.

There seems to be a issue with the file name or directory reference. I seem to get an extra "/" in the path description to the input photos. See below example.

'/home/rb/Downloads/ODM3/Pics//DJI00829.jpg'

How can I get around this to complete the processing?

Many thanks.
RB

RB-RSA, 2015-04-16:
Managed to remove the offending "/" but still get the same error. 

Here is the full description below.

  - running texturing - Thu Apr 16 21:57:07 SAST 2015

Error in OdmTexturing:
Error in OdmTexturing:
Failed to read image:
'/home/rb/Downloads/ODM3/Pics/DJI00829.jpg'

For more detailed information, see log file.

quitting cause: 
	"/home/rb/Downloads/ODM3/Pics/../bin/odm_texturing" -bundleFile /home/rb/Downloads/ODM3/Pics/reconstruction-with-image-size-2400/pmvs/bundle.rd.out -imagesPath /home/rb/Downloads/ODM3/Pics -imagesListPath /home/rb/Downloads/ODM3/Pics/reconstruction-with-image-size-2400/pmvs/list.rd.txt -inputModelPath /home/rb/Downloads/ODM3/Pics/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -outputFolder /home/rb/Downloads/ODM3/Pics/reconstruction-with-image-size-2400-results/odm_texturing/ -textureResolution 4096 -bundleResizedTo 2400 -textureWithSize 3600 -logFile /home/rb/Downloads/ODM3/Pics/reconstruction-with-image-size-2400/odm_texturing/odm_texturing_log.txt
returned with code 256

smathermather, 2015-04-16:
Rename all images to .jpg from .JPG. This is a known bug.

smathermather, 2015-04-16:
And the original bug was reported here: https://github.com/OpenDroneMap/OpenDroneMap/issues/33

closed 2015-04-16 by smathermather

RB-RSA, 2015-04-17:
Thanks a million. Will try this.