OpenDroneMap/ODM #99: Failed to initialize libdc1394

smullero, 2015-04-21:
Hi,

I'm new to OpenDroneMap and I wanted to run a test dataset. I succeeded to install opendronemap but when I ran the script in cd ~/BundlerTools/src/bundler/examples/kermit
~/BundlerTools/./run.pl
I got an error saying failed to initialize libdc1394.
In full:

libdc1394 error: Failed to initialize libdc1394
Error in Georef:
Failed opening /home/vagrant/BundlerTools/src/bundler/examples/kermit/gcp_list.txt for reading.

quitting cause: 
	"/home/vagrant/BundlerTools/bin/odm_georef" -bundleFile /home/vagrant/BundlerTools/src/bundler/examples/kermit/reconstruction-with-image-size-2400/pmvs/bundle.rd.out -gcpFile /home/vagrant/BundlerTools/src/bundler/examples/kermit/gcp_list.txt -imagesPath /home/vagrant/BundlerTools/src/bundler/examples/kermit/ -imagesListPath /home/vagrant/BundlerTools/src/bundler/examples/kermit/reconstruction-with-image-size-2400/pmvs/list.rd.txt -bundleResizedTo 2400 -inputFile /home/vagrant/BundlerTools/src/bundler/examples/kermit/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj -outputFile /home/vagrant/BundlerTools/src/bundler/examples/kermit/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj -logFile /home/vagrant/BundlerTools/src/bundler/examples/kermit/reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt
returned with code 256

What to do to get this to run properly?

bgirardot, 2015-04-21:
Hi,

I am told the libdc1394 error is nothing to worry about and should not affect output. I have run into it in the past in a virtual machine environment when there is no USB emulation. Enabling usb emulation fixed the error for me.

As to the gcp (ground control point) list error, there is a setting to disable using such a file.

--odm_georeferencing-useGcp "false"

Let us know if those help you at all.

There is also a users email list you can join to help you continue to troubleshoot if needed.
http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users

cheers

smullero, 2015-04-21:
Hi,

thanks a lot for the prompt response! Yeah, the results have been produced
despite the error. I'm just so new to this I didn't even realize to check
it. This helped. I also subscribed, it's a very nice product you have here.

Cheers,

Samuli

jewelthief, 2015-07-27:
I wonder libdc1394 is not required (i.e. it does not effect the output), can we just remove its call?

smathermather, 2015-08-07:
@jewelthief -- sure! Do you want to try a pull request?

jewelthief, 2015-09-15:
I just tried this again with latest build and saw libdc1394 error.

darkpioneer, 2015-09-15:
I had the same error, simpe fix "sudo ln /dev/null /dev/raw1394"

jewelthief, 2015-09-15:
Hi Darkpioneer, thanks will try. Below is what I am seeing before running your command

Error in OdmExtractUtm:
No valid GPS position found
For more detailed information, see log file.
libdc1394 error: Failed to initialize libdc1394
Error in Georef:

I do see I got .obj file which opens okay on Meshlab but no orthophoto.. is that normal? Is there a way I can still generate Orthophoto using these pics?

dakotabenjamin, 2015-09-16:
If the photos don't have GPS ephemeris in their EXIF data, then you won't be able to generate an orthophoto AFAIK.

smathermather, 2015-09-17:
Yes, if you have no GPS ephemeris, and you have no ground control points, it can not georeference, and thus will not georeference.

edgarriba, 2015-10-30:
is there any way to set this info in a txt file?

smathermather, 2015-11-01:
@edgarriba -- set the ground control, or the GPS ephemeris? The description of how to set the ground control is here: https://github.com/OpenDroneMap/OpenDroneMap/wiki/2.-Running-OpenDroneMap

A text file of GPS positions for the camera won't work -- it does need to be set in the exif. Mapillary has good tools for this, see: https://github.com/mapillary/mapillary_tools/tree/master/python

edgarriba, 2015-11-01:
nice ! Thanks for the info

closed 2016-10-12 by dakotabenjamin