OpenDroneMap/ODM #104: OpenDroneMap not recognizing "Garmin VIRB"

GeospatialDaryl, 2015-04-27:
Hi-

I've been wrestling with this one for a few weeks - I have the "Make" "Model" EXIF tags set, such that the concatenation in the ccd_defs is returning "Garming VIRB".  But I'm still getting the 'no CCD' error.

From looking at the forums, it seems that I've got non-compliant EXIF somewhere that's confusing on the the early steps in the toolchain (Blender?).

What package should I look at to discern the expected EXIF tags for the processor?   Can anyone provide any insights?

 Then I can compare the working sensors of the same diameter to the complaint-inducing Garmin Virb Elite.

Daryl

smathermather, 2015-04-28:
Hi @GeospatialDaryl -- what do you get when you run ```jhead imagename.jpg```?

smathermather, 2015-04-29:
jhead should give a result similar to this:

```
jhead DSC05889.JPG 
File name    : DSC05889.JPG
File size    : 4471390 bytes
File date    : 2015:04:29 08:26:45
Camera make  : SONY
Camera model : DSC-HX5V
Date/Time    : 2015:04:27 10:42:17
Resolution   : 3648 x 2736
Flash used   : No
Focal length :  4.2mm
Exposure time: 0.0040 s  (1/250)
Aperture     : f/3.5
ISO equiv.   : 250
Whitebalance : Auto
Metering Mode: pattern
Exposure     : program (auto)
GPS Latitude : N 41d 26m 41.735s
GPS Longitude: W 81d 42m 58.986s
GPS Altitude :  183.50m
JPEG Quality : 94
```

Note especially these two lines:
```
Camera make  : SONY
Camera model : DSC-HX5V
```

smathermather, 2015-04-29:
Interestingly, this is what I get when I run ```exif```:

```
exif DSC05889.JPG 
EXIF tags in 'DSC05889.JPG' ('Intel' byte order):
--------------------+----------------------------------------------------------
Tag                 |Value
--------------------+----------------------------------------------------------
Image Description   |                               
Manufacturer        |SONY
Model               |DSC-HX5V
Orientation         |Top-left
X-Resolution        |72
Y-Resolution        |72
Resolution Unit     |Inch
Date and Time       |2015:04:27 10:42:17
YCbCr Positioning   |Co-sited
PRINT Image Matching|28 bytes undefined data
Compression         |JPEG compression
Manufacturer        |SONY
Model               |DSC-HX5V
Orientation         |Top-left
X-Resolution        |72
Y-Resolution        |72
Resolution Unit     |Inch
Date and Time       |2015:04:27 10:42:17
Exposure Time       |1/250 sec.
F-Number            |f/3.5
Exposure Program    |Normal program
ISO Speed Ratings   |250
Exif Version        |Exif Version 2.21
Date and Time (Origi|2015:04:27 10:42:17
Date and Time (Digit|2015:04:27 10:42:17
Components Configura|Y Cb Cr -
Compressed Bits per | 4
Exposure Bias       |0.00 EV
Maximum Aperture Val|3.62 EV (f/3.5)
Metering Mode       |Pattern
Light Source        |Unknown
Flash               |Flash did not fire, compulsory flash mode
Focal Length        |4.2 mm
Maker Note          |9992 bytes undefined data
FlashPixVersion     |FlashPix Version 1.0
Color Space         |sRGB
Pixel X Dimension   |3648
Pixel Y Dimension   |2736
File Source         |DSC
Scene Type          |Directly photographed
Custom Rendered     |Normal process
Exposure Mode       |Auto exposure
White Balance       |Auto white balance
Scene Capture Type  |Standard
Contrast            |Normal
Saturation          |Normal
Sharpness           |Normal
GPS Tag Version     |2.2.0.0
North or South Latit|N
Latitude            |41, 26, 41.735
East or West Longitu|W
Longitude           |81, 42, 58.986
Altitude Reference  |Sea level
Altitude            |183.5
GPS Time (Atomic Clo|11:20:33.00
GPS Receiver Status |V
GPS Measurement Mode|3
Speed Unit          |K
Speed of GPS Receive|1.0
Reference for direct|T
Direction of Movemen|136.03
GPS Image Direction |M
GPS Image Direction |192.50
Geodetic Survey Data|WGS-84
GPS Date            |2015:04:21
GPS Differential Cor|0
Interoperability Ind|R98
Interoperability Ver|0100
--------------------+----------------------------------------------------------
EXIF data contains a thumbnail (14857 bytes).
```

GeospatialDaryl, 2015-05-05:
Hi Stephen - Thanks for the follow up - I'll address this from home later this evening.   I was on some serious Va-cay!

D

smathermather, 2015-05-06:
Excellent. Yes -- good to turn off completely for a time. Welcome back.

smathermather, 2015-11-23:
Hey @GeospatialDaryl -- I'm going to close this. Let me know if it needs re-opened or addressed in any way.

closed 2015-11-23 by smathermather

bearyjd, 2017-12-30:
I am having problems with Garmin Virb EXIF data as well.  Whenever I run WebODM with VIRB photos, I get the following:

--------------------+----------------------------------------------------------
2017-12-30 19:27:23,801 Ceres Solver Report: Iterations: -2, Initial cost: -1.000000e+00, Final cost: -1.000000e+00, Termination: FAILURE
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/reconstruct.py", line 21, in run
    report = reconstruction.incremental_reconstruction(data)
  File "/code/SuperBuild/src/opensfm/opensfm/reconstruction.py", line 1069, in incremental_reconstruction
    data, graph, reconstruction, remaining_images, gcp)
  File "/code/SuperBuild/src/opensfm/opensfm/reconstruction.py", line 1031, in grow_reconstruction
    align.align_reconstruction(reconstruction, gcp, data.config)
  File "/code/SuperBuild/src/opensfm/opensfm/align.py", line 14, in align_reconstruction
    res = align_reconstruction_similarity(reconstruction, gcp, config)
  File "/code/SuperBuild/src/opensfm/opensfm/align.py", line 56, in align_reconstruction_similarity
    return align_reconstruction_naive_similarity(reconstruction, gcp)
  File "/code/SuperBuild/src/opensfm/opensfm/align.py", line 80, in align_reconstruction_naive_similarity
    T = tf.superimposition_matrix(X.T, Xp.T, scale=True)
  File "/code/SuperBuild/src/opensfm/opensfm/transformations.py", line 1041, in superimposition_matrix
    scale=scale, usesvd=usesvd)
  File "/code/SuperBuild/src/opensfm/opensfm/transformations.py", line 954, in affine_matrix_from_points
    u, s, vh = numpy.linalg.svd(numpy.dot(v1, v0.T))
  File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 1359, in svd
    u, s, vt = gufunc(a, signature=signature, extobj=extobj)
  File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 99, in _raise_linalgerror_svd_nonconvergence
    raise LinAlgError("SVD did not converge")
numpy.linalg.linalg.LinAlgError: SVD did not converge
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 122, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
--------------------+----------------------------------------------------------

EXIF tags in 'VIRB0001-68.JPG' ('Intel' byte order):
--------------------+----------------------------------------------------------
Tag                 |Value
--------------------+----------------------------------------------------------
Image Description   |DCIM\100_VIRB\VIRB0001-
Manufacturer        |Garmin
Model               |VIRB Ultra 30
Orientation         |Top-left
X-Resolution        |72
Y-Resolution        |72
Resolution Unit     |Inch
Software            |4.40
Date and Time       |2017:12:29 15:50:38
YCbCr Positioning   |Centered
Compression         |JPEG compression
X-Resolution        |72
Y-Resolution        |72
Resolution Unit     |Inch
Exposure Time       |1/543 sec.
F-Number            |f/2.6
Exposure Program    |Normal program
ISO Speed Ratings   |100
Exif Version        |Exif Version 2.3
Date and Time (Origi|2017:12:29 15:50:38
Date and Time (Digit|2017:12:29 15:50:38
Components Configura|- Cr Cb Y
Compressed Bits per |15561472.087
Shutter Speed       |9.09 EV (1/543 sec.)
Aperture            |2.76 EV (f/2.6)
Exposure Bias       |0.00 EV
Maximum Aperture Val|2.76 EV (f/2.6)
Subject Distance    |0.0 m
Metering Mode       |Center-weighted average
Light Source        |Unknown
Flash               |No flash function
Focal Length        |2.7 mm
Maker Note          |256 bytes undefined data
FlashPixVersion     |Unknown FlashPix Version
Color Space         |sRGB
Pixel X Dimension   |4000
Pixel Y Dimension   |3000
Exposure Index      |0/0
Sensing Method      |One-chip color area sensor
File Source         |DSC
Scene Type          |Directly photographed
Custom Rendered     |Normal process
Exposure Mode       |Auto exposure
White Balance       |Auto white balance
Digital Zoom Ratio  | 1
Focal Length in 35mm|15
Scene Capture Type  |Landscape
Gain Control        |Normal
Contrast            |Normal
Saturation          |Normal
Sharpness           |Normal
Device Setting Descr|4 bytes undefined data
Subject Distance Ran|Unknown
North or South Latit|N
Latitude            |40, 32, 56.169895
East or West Longitu|W
Longitude           |75, 58, 39.558057
Altitude Reference  |Sea level
Altitude            |142.470336
Speed Unit          |K
Speed of GPS Receive|0.000000
Direction of Movemen|243.863808
Geodetic Survey Data|WGS-84
Interoperability Ind|
Interoperability Ver|0010
--------------------+----------------------------------------------------------
EXIF data contains a thumbnail (3370 bytes).

[image]