OpenDroneMap/ODM #106: PCL failing to build 'does not support x86-64 instruction set'

gmaclennan, 2015-05-12:
I am trying to build on a Digital Ocean Ubuntu 14.04 64bit droplet. `./install.sh` fails at pcl.

```sh
#cat logs/pcl_2_build.log

[  1%] Building CXX object common/CMakeFiles/pcl_common.dir/src/point_types.cpp.o
/root/OpenDroneMap/src/pcl/common/src/point_types.cpp:1:0: error: CPU you selected does not support x86-64 instruction set
 /*
 ^
make[2]: *** [common/CMakeFiles/pcl_common.dir/src/point_types.cpp.o] Error 1
make[1]: *** [common/CMakeFiles/pcl_common.dir/all] Error 2
```

Some potentially useful information:

```sh
#uname -a
Linux ddem 3.13.0-24-generic #46-Ubuntu SMP Thu Apr 10 19:11:08 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
```

Virtual machine size is 8GB 4CPU 20GB SSD.

gmaclennan, 2015-05-14:
An update: this worked fine on Ubuntu 14.04 32bit on an otherwise identical setup, so this is related to the 64bit architecture for some reason.

smathermather, 2015-05-14:
I have run this on Ubuntu 14.04 64bit many times (am at the moment, for that matter), so this is a really strange error to get.

Here's my ```uname -a```:

```Linux name-of-my-machine 3.13.0-43-generic #72-Ubuntu SMP Mon Dec 8 19:35:06 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux```

gmaclennan, 2015-05-14:
I found [this discussion](http://forum.openframeworks.cc/t/failure-compiling-of-on-an-ubuntu-12-04-03-lts-digital-ocean-droplet/13893/2) of a similar issue and [this PR](https://github.com/openframeworks/openFrameworks/pull/2723) which solves it, but I don't understand enough (anything!) about compiler flags to understand how this applies to ODM. From the thread it sounds like how the CPU virtualization by xen is mis-reporting architecture on Digital Ocean in some way.

smathermather, 2015-05-15:
Which "Region" are you working in? I've only run Ubuntu 14.04 64-bit Digital Ocean in New York.

Excellent link. Yes, this sounds related. Maybe @dustymugs has some ideas here (to pull him randomly in)... .

dustymugs, 2015-05-15:
You should be able to add the following right before line 390...

```
make install NO_PLATFORM_OPTIMIZATIONS
```

or

```
make PLATFORM_VARIANT=xenkernel
```

smathermather, 2015-05-15:
Awesome. Thanks @dustymugs. Do you want to try that @gmaclennan?

@dustymugs -- in general, any serious performance implications for making ```NO_PLATFORM_OPTIMIZATIONS``` default in the repo?

dustymugs, 2015-05-15:
Yes. There are performance implications for setting ```NO_PLATFORM_OPTIMIZATIONS```. I wouldn't advise it as a default without some test to see the platform. Some like compiling a tiny C file that results in the same error raised when starting the PCL compilation.

dustymugs, 2015-05-15:
@gmaclennan One thing that can help debugging is if you can post the CMakeCache.txt for the pcl build. It should be found in

BASE_PATH_OF_OpenDroneMap/src/pcl/pcl_build/CMakeCache.txt

I hope it's there. I can't find my local installs of PCL at the moment.

smathermather, 2015-05-19:
Hi @gmaclennan -- any luck?

gmaclennan, 2015-05-28:
Sorry about the delay here, got drawn into another project. Unhelpfully, when I tried this again with a new droplet (DigitalOcean NYC region 3) it installed fine without errors. I had deleted my previous test droplet to save money. Maybe DO has a few different virtual machines / different hardware and this was just chance? I will let you know if I come across this again.

gmaclennan, 2015-05-28:
Sorry, just realized something when I checked the `uname -a` from my initial post. I believe that I initially tried this one a testing droplet that I had installed using the DO dokku droplet, which is Ubuntu 14.04 64bit, but perhaps with a different kernal / config. On my current droplet (which works, and is a from-scratch Ubuntu 14.04 64bit) `uname -a` gives:

```
Linux ODM 3.13.0-52-generic #85-Ubuntu SMP Wed Apr 29 16:44:17 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

Anyway, resolved for now. Moral: don't use the Digital Ocean Dokku droplet for ODM (can't imagine why anybody else would do this, but maybe they're just not paying attention like me and trying to recycle droplets to save time / money)

smathermather, 2015-05-29:
Thanks @gmaclennan. Good to have this issue brought up because we'll run into again with hosted services... .  Thanks for following up. (and thanks @dustymugs).

closed 2015-05-29 by smathermather