OpenDroneMap/ODM #108: Computed GSD, Height or Camera distance?

rickesh, 2015-05-12:
Where are the GSD (Ground sample distance) computed in Opendronemap? or is not computed at all?
And also I am interested in knowing from what height the images were taken(depth estimation) from the images. And also I would like to know the baseline distances between the images. Is it stored somewhere?

dbrgn, 2015-09-20:
I'd also be interested in this issue.

dakotabenjamin, 2015-09-21:
Hmm... Could you get the height from the altitude EXIF datum?

smathermather, 2015-11-23:
I am unclear what the question is? Height / depth are implicit in the generation of camera positions and point cloud, so therefore at the bundler step in the traditional pipeline, and inside of OpenSfM in that toolchain. Maybe @paulinus could answer this better for OpenSfM.

paulinus, 2015-11-24:
The GSD formula is correct. You can find the focal length in the file `reconstruction-with-image-size-XXX/opensfm/reconstruction.json`. In there, there is a list of cameras with their estimated `focal`. Note that this value is normalized and corresponds already to focal/width.

The height is not computed anywhere at the moment. We should define with respect to what reference altitude we would want to compute that.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/computing-gsd/376

closed 2018-03-28 by dakotabenjamin