OpenDroneMap/ODM #109: cleanup installation script, decouple from building dependencies.

cezio, 2015-05-15:
Hi,

I have some thoughts on code organization part of ODM:

While installation script works, and creates usable binaries, whole building process could be improved:
 * installation should not expect specific os version (Ubuntu 12.04 is not the freshest one, also there are other distributions), this also means that installation within vm container is not always possible or desired,
 * installation should rely on system-provided packages where possible, and utilize provided tools for configuration (for example, ``pkg-config``), 
 * dependencies should be cleaned up:
   * bundler sources are available on github, from where they can be downloaded as zip: https://github.com/snavely/bundler_sfm/archive/master.zip),
   * parallel, clapack, pcl (and it's dependency: vtk) can be installed as system package in most cases (pcl is available in Ubuntu 15.04). Especially important with pcl/vtk, which takes some time to compile.
   * vfleat newest version is 0.9.20: (http://www.vlfeat.org/download/vlfeat-0.9.20-bin.tar.gz), it can be also found in Ubuntu 14.04 packages,
   * cmvs is added as a download link, but also is added as a blob and concated to regular archive during installation, which suppress archive download.
 * installation should be split into phases - dependency check and configuration, actual build and binaries installation,
 * if possible, build process should be called with standard build tools (make/cmake), since parts of it are using it already. This brings back previous point, because It seems build configuration part lacks dependency autodiscovery.

 I think ultimately this should lead to more maintenance-friendly build procedure, so actual build would be minimal, and ODM could be packaged and installed as a regular package.

cezio, 2015-05-15:
One more thing here, running ``install.sh`` script will remove all previous build artifacts from build outputs. I think build script should be able to work incrementally here.

cezio, 2015-05-15:
Some work in this direction: https://github.com/OpenDroneMap/OpenDroneMap/issues/109

dakotabenjamin, 2015-08-17:
I was looking at install.sh and noticed something:
 * some dependencies are included in the repo and are included in the "getting the sources" section: vlfeat, cmvs parts, PoissonRecon, graclus, and clapack. For ease of updating the sources in the future, they should be included in .gitignore.

dakotabenjamin, 2016-09-12:
We use cmake now!

closed 2016-09-12 by dakotabenjamin

smathermather, 2016-09-13:
w00t!