OpenDroneMap/ODM #112: libdc1394 error: Failed to initialize libdc1394

joseluisbossio, 2015-05-22:
Hi, we have a problem to testing caliterra example while running texturing:

 - running texturing - Fri May 22 16:08:12 ART 2015

libdc1394 error: Failed to initialize libdc1394
Error in OdmTexturing:
Error in OdmTexturing:
Failed to read image:
'/home/cpacheco/odm_data/caliterra//IMG_9354.jpg'

For more detailed information, see log file.

quitting cause:
        "/home/cpacheco/OpenDroneMap/bin/odm_texturing" -bundleFile /home/cpacheco/odm_data/caliterra/reconstruction-with-image-size-2400/pmvs/bundle.rd.out -imagesPath /home/cpacheco/odm_data/caliterra/ -imagesListPath /home/cpacheco/odm_data/caliterra/reconstruction-with-image-size-2400/pmvs/list.rd.txt -inputModelPath /home/cpacheco/odm_data/caliterra/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -outputFolder /home/cpacheco/odm_data/caliterra/reconstruction-with-image-size-2400-results/odm_texturing/ -textureResolution 4096 -bundleResizedTo 2400 -textureWithSize 3600 -logFile /home/cpacheco/odm_data/caliterra/reconstruction-with-image-size-2400/odm_texturing/odm_texturing_log.txt
returned with code 256

Thanks in advance... 
JL

joseluisbossio, 2015-05-23:
We executed the following command:  sudo ln /dev/null /dev/raw1394
relative the message error: "libdc1394 error: Failed to initialize libdc1394"

After that, the same error on OdmTexturing happened:

Error in OdmTexturing:
Failed to read image:
'/home/cpacheco/odm_data/caliterra//IMG_9354.jpg'

It's happened again with another example called 'benckmark'.

smathermather, 2015-05-23:
Ah. Try renaming to .jpg instead of .JPG.

smathermather, 2015-05-24:
Also, you can ignore the libdc1394 error. It should be a warning.

joseluisbossio, 2015-05-25:
Hi, after rename the extension files to lower case, we have the following
problem:

tail nohup.out

Error in Georef:

Failed opening /home/cpacheco/odm_data/benchmark/gcp_list.txt for reading.

What about the gpc_list.txt file?

quitting cause:
  "/home/cpacheco/OpenDroneMap/bin/odm_georef" -bundleFile
/home/cpacheco/odm_data/benchmark/reconstruction-with-image-size-2400/pmvs/bundle.rd.out
-gcpFile /home/cpacheco/odm_data/benchmark/gcp_list.txt -imagesPath
/home/cpacheco/odm_data/benchmark/ -imagesListPath
/home/cpacheco/odm_data/benchmark/reconstruction-with-image-size-2400/pmvs/list.rd.txt
-bundleResizedTo 2400 -inputFile
/home/cpacheco/odm_data/benchmark/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj
-outputFile /home/cpacheco/odm_data/benchmark/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj
-logFile /home/cpacheco/odm_data/benchmark/reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt
returned with code 256

Thanks in advance!
JL

smathermather, 2015-05-25:
Yes, there is no georeference file included for that dataset, so that is just telling you that. Cheers! It completed!

smathermather, 2015-05-25:
(That is an optional file that allows for georeferencing and orthophoto generation).

joseluisbossio, 2015-05-26:
;) thank buddy!

About the gcp_list.txt, we are very interested in orthophotos. Do you have
(public domain) documentation about how to create this kind of photos and
the coordinates file?

smathermather, 2015-05-26:
NP, @joseluisbossio. Check the wiki for documentation on gcp_list.txt. It's pretty simple, but let me know if any clarification is needed:

https://github.com/OpenDroneMap/OpenDroneMap/wiki/Running-OpenDroneMap

closed 2015-05-26 by smathermather

jewelthief, 2015-07-24:
I am getting "libdc1394 error: Failed to initialize libdc1394" but after looking at above comment by smarthermather that its a warning, i have kept the processing job ON. but its not showing any new update for more than an hour now. Not sure if i have to kill the job and start again....please guide

jewelthief, 2015-07-24:
Update: I see the results folder have been created, but obj file is missing in texturing folder. How do i avoid this?

smathermather, 2015-08-07:
Looks like this got resolved by letting it run to completion. Can you confirm @jewelthief? Reopen if you need to.

jewelthief, 2015-08-07:
We can close this. Thanks @smathermather

smathermather, 2015-08-08:
Cool. Thanks @jewelthief