OpenDroneMap/ODM #114: tar.gz of output

smathermather, 2015-06-05:
Referencing @dakotabenjamin's fork -- love the tar.gz. The only hitch I see is if I run the process at more than one image resample resolution, the last run over-writes the previous.

e.g.:
```ts run.pl --resize-to 1200```
will be overwritten by the subsequent
```ts run.pl --resize-to 2400```

dakotabenjamin, 2015-06-09:
yeah, reconstr-results.tar.gz was a placeholder name. Has since been updated. Right now the only weird thing about the code is that the archive is 5 folders deep from root. i.e /home/dmb2/github/odm_data/odm_data_seneca/reconstruction-with-image-size-2400-results/. This is due to the use of the term $jobOptions{jobDir}-results/*, which returns the whole address.

smathermather, 2015-06-10:
Awesome. Ya, the whole address is an issue, as it exposes a username.

closed 2015-08-26 by dakotabenjamin