OpenDroneMap/ODM #115: PMVS step does not include all output point clouds

smathermather, 2015-06-07:
Issue is here:
https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/run.pl#L730

petrasovaa, 2015-06-20:
To better describe the problem, this line:

    run("\"$BIN_PATH/pmvs2\" pmvs/ option-0000");

processes only 1 file, but there can be more, depending on the number of input images. Cmvs creates pmvs.sh which can be called instead, that would be easy to implement. However, then the meshing step accepts only one file anyway. So ideally it would read all the point clouds and create one mesh.

For now, we can set parameter `--cmvs-maxImages` to higher number (default is now 100) which ensures only one point cloud (while loosing the possibility to parallelize pmvs step).

smathermather, 2015-06-21:
Hmm, good point on the work around. Thanks @petrasovaa! Do you mind doing a pull request for this, and then we'll reduce that size (--cmvs-maxImages) during the python port.

smathermather, 2016-12-20:
I think this is resolved, yes @dakotabenjamin

dakotabenjamin, 2016-12-21:
More or less, also with `--use-opensfm-pointcloud` it becomes less relevant

closed 2016-12-21 by dakotabenjamin

smathermather, 2016-12-21:
And I think we fixed this in an iteration of the python port. Awesome.