OpenDroneMap/ODM #119: odm_orthphoto.png is only 1x1 px

wolkstein, 2015-06-17:
hi,
i run ODM successfully with a ok result. my odm_textured_model_geo.obj is finally created also all needed textures.
[image]

but the odm_orthphoto.png is only 1x1 px.
here is the foto:
[image]
it is not visible because it is only 1x1 px big.

i run the script like this:
```
perl /home/wolke/develop/git/OpenDroneMap/run.pl --resize-to 1500 --odm_georeferencing-useGcp "true"
```
the script finished without any error message.

here is my odm_orthophoto_log file:
```
/home/wolke/develop/git/OpenDroneMap/bin/odm_orthophoto_log.txt

Arguments given
-inputFile
/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-1500-results/odm_texturing/odm_textured_model_geo.obj
-logFile
/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-1500/odm_orthophoto/odm_orthophoto_log.txt
-outputFile
/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-1500-results/odm_orthphoto.png
-resolution
20.0

Reading textured mesh from: /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-1500-results/odm_texturing/odm_textured_model_geo.obj
Log file path was set to: /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-1500/odm_orthophoto/odm_orthophoto_log.txt
Writing output to: /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-1500-results/odm_orthphoto.png
Resolution count was set to: 20pixels/meter

Reading mesh file...
.. mesh file read.

Number of faces in the model 199911
Set boundary to contain entire model.
Local boundary points:
Point 1: 0.195994 0.726776
Point 2: 0.195994 0.72886
Point 3: 0.198091 0.72886
Point 4: 0.198091 0.726776

Ortho photo bounds x : 0.195994 -> 0.198091
Ortho photo bounds y : 0.726776 -> 0.72886
Ortho photo area : 4.37136e-06m2
Ortho photo resolution, width x height : 1x1
Translating and scaling mesh...
.. mesh translated and scaled.

Rendering the ortho photo...
Warning: Sliver polygon found at face index 0

!!this warning happens for each face from 0 - 199872. i remove all lines to make it possible to post the log file here!!

Warning: Sliver polygon found at face index 199872
Material 11 rendered.
...ortho photo rendered

Writing ortho photo to /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-1500-results/odm_orthphoto.png
Orthophoto generation done.

```
any idea what goes wrong?

i also get an 1x1 px ortho photo if i successfully build(run.pl run all steps without error) the  odm_data_copr example from this github project.

/g
wolke

smathermather, 2015-06-17:
Hi,

Can you please post this to the OpenDroneMap users group? Thanks for the very good detail. Go ahead and link to this issue.

http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users

Thanks

wolkstein, 2015-06-19:
hi,
after some test runs with only odm_orthophoto binary like this:
```
/home/wolke/develop/git/OpenDroneMap/bin/odm_orthophoto -inputFile reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj -logFile odm_orthophoto_log.txt -outputFile odm_orthphoto.png -resolution 800.0
```
i successfully created an orthophoto.
the different was that i create the orthophoto manually from the odm_textured_model.obj file and not from the odm_textured_model_geo.obj file. i noticed that the odm_textured_model.obj only produce some silver polygon warnings instead of the warnings from odm_textured_model_geo.obj. in this file each polygon was warned from odm_orthophoto creation. it looks like that the geo referencing procedure do not clean. also i noticed that bounds are completely different.

output from odm_orthophoto if odm_textured_model.obj is used:
```
Number of faces in the model 347970
Set boundary to contain entire model.
Local boundary points:
Point 1: -8.7477 -6.5398
Point 2: -8.7477 4.2371
Point 3: 2.0291 4.2371
Point 4: 2.0291 -6.5398

Ortho photo bounds x : -8.7477 -> 2.0291
Ortho photo bounds y : -6.5398 -> 4.2371
Ortho photo area : 116.141m2
Ortho photo resolution, width x height : 8622x8622
Translating and scaling mesh...
.. mesh translated and scaled.
```
output from odm_orthophoto if odm_textured_model_geo.obj is used:
```
Number of faces in the model 347970
Set boundary to contain entire model.
Local boundary points:
Point 1: 0.195994 0.726759
Point 2: 0.195994 0.728876
Point 3: 0.198113 0.728876
Point 4: 0.198113 0.726759

Ortho photo bounds x : 0.195994 -> 0.198113
Ortho photo bounds y : 0.726759 -> 0.728876
Ortho photo area : 4.48494e-06m2
Ortho photo resolution, width x height : 1x1
Translating and scaling mesh...
.. mesh translated and scaled.
```

both Ortho photo area calculations are completely wrong but it looks like that the calculated area in  odm_textured_model_geo.obj is completely to low.   4.48494e-06m2 is something you can only see with an microscope:). maybe this cased the 1x1 pixel result.

/g
wolke

ps:
i publish this obj model on sketchfab. https://sketchfab.com/models/6cbaa92880424f0c96950dd451d41f98

wolkstein, 2015-06-19:
i am sure that every one can reproduce this with odm_data_copr example data set.
/g
wolke

wolkstein, 2015-06-20:
please can you reopen this issue. imo there is a bug in odm_referencing process if gcp_txt is used.

here is my odm_georeferencing_log.txt:
```
/home/wolke/develop/git/OpenDroneMap/bin/odm_georef_log.txt
Arguments given
-bundleFile
/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400/pmvs/bundle.rd.out
-gcpFile
/home/wolke/Bilder/opendronemap-test/loehne-teich/gcp_list.txt
-imagesPath
/home/wolke/Bilder/opendronemap-test/loehne-teich/
-imagesListPath
/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400/pmvs/list.rd.txt
-bundleResizedTo
2400
-inputFile
/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj
-outputFile
/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj
-logFile
/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt

Reading cameras from: /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400/pmvs/bundle.rd.out
Reading GCPs from: /home/wolke/Bilder/opendronemap-test/loehne-teich/gcp_list.txt
Images location is set to: /home/wolke/Bilder/opendronemap-test/loehne-teich/
Reading image list from: /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400/pmvs/list.rd.txt
Bundle resize value is set to: 2400
Reading textured mesh from: /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj
Writing output to: /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj
Log file path was set to: /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt

Reading following GCPs from file:
x_: 52.1972 y_: 8.72714 z_: 0 pixelX_: 1364 pixelY_: 1953 image: p1100149.jpg
x_: 52.1971 y_: 8.72745 z_: 0 pixelX_: 2522 pixelY_: 3646 image: p1100149.jpg
x_: 52.1966 y_: 8.72739 z_: 0 pixelX_: 2069 pixelY_: 447 image: p1100206.jpg
x_: 52.1966 y_: 8.7278 z_: 0 pixelX_: 1777 pixelY_: 2364 image: p1100206.jpg
x_: 52.1966 y_: 8.72739 z_: 0 pixelX_: 1514 pixelY_: 622 image: p1100207.jpg
x_: 52.1966 y_: 8.7278 z_: 0 pixelX_: 1202 pixelY_: 2543 image: p1100207.jpg
x_: 52.1972 y_: 8.72805 z_: 0 pixelX_: 1136 pixelY_: 1147 image: p1100224.jpg
x_: 52.1971 y_: 8.72745 z_: 0 pixelX_: 1394 pixelY_: 3698 image: p1100224.jpg
x_: 52.1966 y_: 8.7278 z_: 0 pixelX_: 1803 pixelY_: 2356 image: p1100232.jpg
x_: 52.1966 y_: 8.72739 z_: 0 pixelX_: 2141 pixelY_: 4313 image: p1100232.jpg
x_: 52.1966 y_: 8.7278 z_: 0 pixelX_: 1969 pixelY_: 1380 image: p1100237.jpg
x_: 52.1972 y_: 8.72714 z_: 0 pixelX_: 724 pixelY_: 3585 image: p1100251.jpg

The calculated easting offset for the georeferenced system: 52
The calculated northing offset for the georeferenced system: 8

Recalculated GCPs with offset:
x_: 0.197234 y_: 0.727137 z_: 0
x_: 0.197133 y_: 0.727449 z_: 0
x_: 0.196568 y_: 0.727388 z_: 0
x_: 0.196604 y_: 0.727803 z_: 0
x_: 0.196568 y_: 0.727388 z_: 0
x_: 0.196604 y_: 0.727803 z_: 0
x_: 0.197214 y_: 0.728053 z_: 0
x_: 0.197133 y_: 0.727449 z_: 0
x_: 0.196604 y_: 0.727803 z_: 0
x_: 0.196568 y_: 0.727388 z_: 0
x_: 0.196604 y_: 0.727803 z_: 0
x_: 0.197234 y_: 0.727137 z_: 0

Reading mesh file /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj

Successfully loaded /home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj.
Position in model for gcp 1: x=-3.35416 y=1.28964 z=-4.7991
Position in model for gcp 3: x=-4.40087 y=-1.91542 z=-5.10136
Position in model for gcp 4: x=-6.11225 y=-2.1928 z=-5.45103
Position in model for gcp 5: x=-5.21149 y=-1.47096 z=-5.25583
Position in model for gcp 6: x=-6.93156 y=-1.77974 z=-5.41621
Position in model for gcp 7: x=-1.79677 y=-2.2021 z=-4.68408
Position in model for gcp 9: x=-4.78418 y=-2.64651 z=-5.05699
Position in model for gcp 11: x=-4.72855 y=-2.62407 z=-5.04944

Choosing optimal gcp triplet...
Mean georeference error 4.48195e-05
Optimal gcp triplet chosen: 3, 4, 6

Final transform:
[ 0.00013678,	0.00005668,	0.00002707,	0.19771138 ]
[ 0.00005703,	-0.00013925,	0.00000339,	0.72786469 ]
[ 0.00002632,	0.00000718,	-0.00014802,	-0.00063022 ]
[ 0.00000000,	0.00000000,	0.00000000,	1.00000000 ]

Applying transform to mesh...
.. mesh transformed.

Successfully saved model.

Saving georeference system file to '/home/wolke/Bilder/opendronemap-test/loehne-teich/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo_georef_system.txt'...
... georeference system saved.

```

/g
wolke

smathermather, 2015-06-20:
I will reopen when I get back to Internet. Thanks for running this down, @wolke.

smathermather, 2015-11-23:
@wolkstein -- can you test with the latest branch?

closed 2016-10-12 by dakotabenjamin