OpenDroneMap/ODM #126: can't run OpenDroneMap

leohardtke, 2015-07-17:
Hi there!
This is a realy nice project!!! I was trying to run in on the benchmark folder as a first attempt, before trying with my own images, but i got: 

leo@leo-Aspire-4830T ~/ap/odm_data/benchmark $ ../../OpenDroneMap/run.pl 
, or } expected while parsing object/hash, at character offset 11081 (before ""Mantis i23": 45.00\n...") at ../../OpenDroneMap/run.pl line 36, <$fh> chunk 1.

The install script finished succesfully on a Ubuntu 14.04 box!
Any idea? Help is appreciated!

smathermather, 2015-07-19:
Go ahead and post to the mailing list. Cheers! Best, Steve.

LeithHawkins, 2015-07-20:
@leohardtke  did you get a result from the mailing list ?  I am having the same issue on 15.04 ?   am about to post on stack exchange but will wait to see if you got an answer

leohardtke, 2015-07-27:
No @LeithHawkins, I didn't post on the mailing list... I'm having busy days at work right now... 
Hope you get an answer!

dbrgn, 2015-08-08:
Are you running `run.pl` from the original directory, or did you move / symlink it to somewhere else?

leohardtke, 2015-08-20:
Hi, I made a new install and now it's working... Thanks for answering and sorry for the noise!

closed 2015-08-20 by leohardtke