OpenDroneMap/ODM #127: Multithreaded odm_texturing?

FLYBYME, 2015-07-18:
Im finding the the `running texturing` step seems to only use one core on my 6 core cpu.

would it ever become multithreaded?

smathermather, 2015-07-19:
Not a bad idea. AFAIK, no one has specific plans, but we accept pull requests :D.

Texturing is a fraction of the total processing time in the datasets I've run. Are you hitting a bottleneck with texturing?

FLYBYME, 2015-07-20:
I would not know where to start with it. Im just learning it now.

This it would be a good idea.

smathermather, 2015-07-21:
It's a great idea, @FLYBYME. I'll add it to the list of feature requests.

smathermather, 2020-04-14:
I will close this for now. If someone feels an urge to chase this or update this issue, we can re-open.

closed 2020-04-14 by smathermather