OpenDroneMap/ODM #128: odm_texturing path error

venox7, 2015-07-20:
Hi,

I'm trying to run OpenDroneMap for the first time on a freshly installed and fully upgraded Ubuntu 15.04.

Everything goes well until odm_texturing starts.  From the log file it seems that the file path its trying to use is incorrect when referring to the images. 

Log file copied here:
Bundle path was set to: /home/venox7/Pictures/35BBarrett/25m/reconstruction-with-image-size-2400/pmvs/bundle.rd.out
Images path was set to: /home/venox7/Pictures/35BBarrett/25m/
Images list path was set to: /home/venox7/Pictures/35BBarrett/25m/reconstruction-with-image-size-2400/pmvs/list.rd.txt
Input model path was set to: /home/venox7/Pictures/35BBarrett/25m/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply
Output folder path was set to: /home/venox7/Pictures/35BBarrett/25m/reconstruction-with-image-size-2400-results/odm_texturing/
The texture resolution was set to: 4096
The resized resolution used in bundler was set to: 2400
The resolution to texture with was set to: 3600
Log file path was set to: /home/venox7/Pictures/35BBarrett/25m/reconstruction-with-image-size-2400/odm_texturing/odm_texturing_log.txt
Successfully loaded 199819 polygons from file.
Successfully read the bundle file.
Successfully read the image list file.
Error in OdmTexturing:
Error in OdmTexturing:
Failed to read image:
'/home/venox7/Pictures/35BBarrett/25m//IMG_2718.jpg'

Is that a coding issue or am I doing something wrong?

Cheers,
v7.

venox7, 2015-07-20:
Just came across the JPG vs jpg issue.  Will try it.

closed 2015-07-20 by venox7

smathermather, 2015-07-21:
Soon to be fixed! Glad you found the issue.

Cheers