OpenDroneMap/ODM #130: Git post-receive hook, or CI check of ccd_defs.json

xxfelixxx, 2015-07-20:
Enhancement Request:

Add a post-receive hook, or add a test to your continuous integration server to check that ccd_defs.json is ok, since it gets edited quite often, and if it is mal-formed, it creates a bad user experience (nothing works).

Suggest something like the following (checking that shell status = 0 [success] )
perl -e 'use JSON qw( decode_json ); open my $fh, "ccd_defs.json"; local $/; my $json = <$fh>; close $fh; my $data = decode_json($json);'

smathermather, 2015-07-21:
Hi @xxfelixxx -- this is a great idea. Would you be willing to do a pull request?

xxfelixxx, 2015-07-21:
Sure.  I pushed patch-2 with some code.

Felix

dakotabenjamin, 2015-08-19:
Added in commit 57b3bfe3e34f77e546be0fd88c38d278a50ad277

closed 2015-08-19 by dakotabenjamin