OpenDroneMap/ODM #131: Distributing run.pl?

FLYBYME, 2015-07-21:
I would like some help to understand how run.pl works.

I have been looking at the code and I think it would be fairly easy to setup some sort of distributed system to run say `KeyMatch` over more then one pc.

From what I can tell everything that is run through `parallel` could be distributed.

Has anyone thought about doing something like this?

GeospatialDaryl, 2015-07-27:
Hi Tim-

I put together a pseudo-code template for run.pl - as well as the skeleton
of a python class definition for the project.  These are the Z.1 and Z.2 on
the github wiki.

In general, developing a more transparent driver script would facilitate
substituting / changing processing steps.  Indeed,  there has been a lot of
interest in including parallel or GPU-enabled processing steps.

My goal in proposing that structure was to initiation a conversation and
collaboration on how to approach that class definition and code
development.  Since then, Stephen has indicated (correct me if I'm wrong)
that there is a commercial firm that will be creating a python
implementation of ODM, leveraging whatever materials have been developed
before.

I am unsure what the best way to collaborate around this workflow - I see a
lot of value to a modular, clean class definition but am a bit
uncomfortable forging that structure on my own.

So - I think there may still be some utility to defining these basic
components, and developing a pseudo-code class definition that spells out
how these methods, and the underlying data models, would be supported.

Is there utility in trying to collaborate about this head of the commercial
development?  Or best to just leave it alone and work from the results?

Thanks-

Daryl

smathermather, 2015-07-28:
At this point, it's probably useful to wait. We are a few short weeks away from the new python control script.

@GeospatialDaryl -- I pointed the Spotscale folks to your pseudo-code, as well as other partial ports and examples, so hopefully they gleaned some insight from your structure. Everyone is on vacation at the moment, so we'll know more soon.

dbrgn, 2015-08-08:
Are you running `run.pl` from the original directory, or did you move / symlink it to somewhere else?

smathermather, 2015-08-24:
Comments, testing, changes, etc. welcome on this new branch: https://github.com/OpenDroneMap/OpenDroneMap/tree/python-port

dbrgn, 2015-08-24:
@smathermather: Here a two quick observations:

- If you want your coding style to be compliant with almost all other Python projects, it should conform to [PEP8](http://legacy.python.org/dev/peps/pep-0008/). This can easily be tested using [flake8](https://github.com/PyCQA/flake8). The most obvious thing right now would be camelCase vs snake_case function names.
- The argument parsing with RegEx looks quite messy. Instead, why not use [argparse](https://docs.python.org/3.4/library/argparse.html) (if you want to stick to the stdlib) or [docopt](https://github.com/docopt/docopt) (if external dependencies are ok)?

smathermather, 2015-08-25:
Good advice, @dbrgn. Looks like based on this comment, @dakotabenjamin has started doing this: https://github.com/OpenDroneMap/OpenDroneMap/pull/158/files

GeospatialDaryl, 2015-09-22:
Hi-

I finished a clean commit of ojbRun.py. In this commit:

   - Increased PEP-8 compliance for line length
   - Created the ODMJob class
      - instantiates to a single job, and __init__s a folder of jpg into
      - ODMPhoto class objects, which contain all metadata in object
      attributes.
   - Factored the first two 'steps'/calls into the ODMJob class as methods
   - Cleaned up some functions and a bit of organization.

The code base is cleaner and easier to read as well.

@ <https://github.com/dakotabenjamin>dakotabenjamin
<https://github.com/dakotabenjamin> - thanks for the feedback yesterday on
the 'unable to merge' - I was having trouble with my local git install.

@smathermather - this commit and code revision picks up on some of the
concepts from this thread.

ODM Dev Community - I strongly feel this OO approach is the way to go - it
facilitates building a GUI and increases the modularity of individual steps
in the processing chain. I will add some additional comments to try to
clarify the class structure - right now, the legacy list properties are
kept using the original handles/names, and then, when parsed, added to the
usually-more-descriptive object attribute name.

The next steps - for which I would like feedback - would include setting up
a pickle approach for keeping track of runs, and re-runs.  Class definition
should be expanded to track this.

Keep flying them kites, and quads, and hot-air balloons-

Daryl

dbrgn, 2015-09-23:
> a pickle approach

Can we use a plain text format like JSON instead? Pickle has security issues, is not easily readable and needs to track the version. JSON is safe and could be modified in case someone wants to play around with the saved state.

GeospatialDaryl, 2015-09-23:
>
> JSON is safe and could be modified in case someone wants to play around
> with the saved state.

Yes - that sounds right.  Also can be versioned in a CVS.  Thanks-

D

dbrgn, 2015-09-23:
We still need some kind of versioning / validation, but as it's not very important data we can simply abort with "invalid state, please remove" when something's not right :)

> Also can be versioned in a CVS.

Are you talking about the version control system? Is that actually still being used? :open_mouth:

GeospatialDaryl, 2015-09-23:
@dbrgn  -

No - I was using the 'Content Version System' as a generic term for any version-aware CMS, no confusion.

How about jsonpickle for storage?

https://jsonpickle.github.io/

Alternately, it looks like most folks use json.dumps() method on object.__dict__ as a serialization.  None of the objects are super complex, but I think jsonpickle looks better.

dbrgn, 2015-09-24:
Personally I'd prefer an explicit format specification and serialization (by constructing a dict from the instance attributes in a `to_json()` method and then serializing it), but if you prefer automatic serialization that would probably be fine too in this case.

dakotabenjamin, 2015-12-09:
@edgarriba has made some significant improvements not only to the installation process but also the OO approach to ODM. Please check out https://github.com/edgarriba/OpenDroneMap.git (the cmake branch)

FLYBYME, 2016-09-20:
I have not used the software in some time now. What is the status of this issue currently?

dakotabenjamin, 2016-09-20:
The process is largely object-oriented code now but still needs some work. It is fully functional and incorporated into the master branch. We use a framework called [ecto](https://github.com/plasmodic/ecto) to help with the processing. We jhaven't looked too much into distribution over multiple PCs yet, I don't think there are any plans to do so.

FLYBYME, 2016-09-21:
You guys are doing amazing work by the way!

Just looking at pmvs.py your just running commands it would not be hard to distribute the commands over many servers.

I dont know python all that much so I cant write code python but I should be able to write something up that can distribute the commands.

Use something like redis as a job queue and nfs as a shared file system. Distributing the commands can give a 10 fold improvement in processing time. I remember back in the day you where using parallel but I dont know if its the same now.

I will have to dig more into the code and see what I can find.

smathermather, 2016-09-22:
Hmm, @dakotabenjamin -- do you think we should close this issue? :)

closed 2016-10-11 by dakotabenjamin