OpenDroneMap/ODM #138: orthophoto.png is not getting created

jewelthief, 2015-07-28:
I have latest bits running on VM with sufficient cpu and memory however in my 2-3 trial run of the program on the sample odm_data shared on this repo, none of them have created orthophoto.png. What could be the potential problem and what log files to look into for the same?

jewelthief, 2015-07-29:
By re-installing OpenDroneMap and re-running the job this time around I was able to generate orthophoto.png, however I had to run "run it with argument --odm_georeferencing-useGcp false --start-with odm_georeferencing" .

Other things I am noticing here is that images output I got from running this process on odm_data_caliterra is different than what Nathan Baldwin is getting on his youtube videos. My output is particularly much more grainy and edgy with surface areas not getting smoothened out (very obvious in the areas marked out). Trying to find what i might have been doing wrong here?

See images attached,

[image]

[image]

dakotabenjamin, 2015-07-29:
I'm wondering if this has to do with texturing changes that came from  #122 and #113. Also, ODM does not do very well in general with vegetation, so you're almost always going to find gaps in those spots.

jewelthief, 2015-07-29:
Well issue seems to be effecting not just vegetation part but also those structures, which are rendered half broken in my output vs Dennis's output.

wolkstein, 2015-08-02:
hi,
i have the same issue since this commit. https://github.com/OpenDroneMap/OpenDroneMap/commit/5b85628e6b71a3e4742d0ab49912ee12622dc7f6 . the texture is completely fragmented.

/g
wolke

dakotabenjamin, 2015-08-04:
Have you tried comparing to the the texture results from before that commit? @geraldmwangi have you noticed any differences being shown here?

wolkstein, 2015-08-05:
i compare a the output from rev:
https://github.com/OpenDroneMap/OpenDroneMap/tree/636f843fd4b314014dc6248007539c85a57515c6
"left site" of the image against rev: 
https://github.com/OpenDroneMap/OpenDroneMap/tree/ebf5e267407c49ab0aa0898f44f28225a872ee72 (currently head)
"right site" of the image.
[image]

the right side (head  rev snapshot) looking more crisp but you also noticed a lot of artefacts.
the  left side ( older revision just two commits before this change set 5b85628 ) looks less sharpen but also less artefacts.

dakotabenjamin, 2015-08-06:
I renamed to issue to something more appropriate to the problem we are facing. 

I've been going over the code changes trying to understand exactly how the algorithm works. The newer code adds two camera selection criteria (as in #113): distance from face and angle between face normal and viewplane normal. What seems to be happening now is that neighbouring triangles are more likely to be textured by a different camera. This may be the cause of the greater distortion and number of artefacts. 

How do we fix this? I'm not an expert in this field, but I have some ideas:

  1. Add the old code back in and add an option to use the new code
  2. Add a new criteria that examines neighbors and prioritizes the cameras they use.
  3. Texture blending

smathermather, 2015-08-07:
Agree with @dakotabenjamin, this may be a premature change without blending / histogram matching, etc..

One option: we could revert for now, as we should see some major texturing changes come down the pike in the next couple of weeks. Alternatively, if @geraldmwangi has some ideas of how to improve on this contribution, that'd be great.

geraldmwangi, 2015-12-18:
Hi I'm sorry I didn't see this thread. I don't have much time now since I'm writting my phd thesis. In a few months I'll get back to this issue. But I have one idea: In line 597 in OdmTexturing::triangleToImageAssignment():
For each face Iook for the camera with the lowest angle& distance to the face. One could instead look at groups of connected faces, say 4 adjacent faces, and assign the camera with the lowest mean angle and mean distance to all 4 faces.

dakotabenjamin, 2016-03-23:
@geraldmwangi look at the new changes from the branch `mvstex` if you are still interested in improving the texturing portion.

closed 2016-10-12 by dakotabenjamin