OpenDroneMap/ODM #144: Georeferencing Orthophoto

dbrgn, 2015-08-09:
I'm using a dataset with embedded GPS coordinates. When generating the orthophoto, it is orthorectified, but I cannot find any georeferencing properties in the resulting PNG image:

```
$ identify -verbose odm_orthphoto.png
...
  Properties:
    date:create: 2015-08-09T04:37:17+02:00
    date:modify: 2015-08-09T04:37:17+02:00
    png:IHDR.bit-depth-orig: 8
    png:IHDR.bit_depth: 8
    png:IHDR.color-type-orig: 2
    png:IHDR.color_type: 2 (Truecolor)
    png:IHDR.interlace_method: 0 (Not interlaced)
    png:IHDR.width,height: 14236, 14143
    png:sRGB: intent=0 (Perceptual Intent)
    signature: a2e742831752696568b1b967d075768b9fff172cc6c8f3630de8dacbc034e6b2
```

Is this only supported with GCPs? Shouldn't this also be possible with coordinates embedded in the metadata?

smathermather, 2015-08-13:
Hi, sorry for the delay. Yes, that is the expected (albeit not ideal behavior). We should have an update soon (this month) that will:

1. Take a proj4 input
2. Output an orthorectified and georeferenced geotiff

In the mean time, the image is orthorectified, but not georeferenced.

seanXYZ, 2015-09-10:
Hi, guys, Does the problem have solved ? A month pasted.
after trying with default parameters , it still runs like this
 
Warning: No GCP file. Consider rerunning with argument --odm_georeferencing-useGcp false --start-with odm_georeferencing
Skipping orthophoto

should I try --odm_georeferencing-useGcp false --start-with odm_georeferencing ?? 
I want try later

smathermather, 2015-09-10:
Hi @seanXYZ -- try this branch: https://github.com/OpenDroneMap/OpenDroneMap/tree/python-port and use the ```run.py``` instead of ```run.pl```. If you don't use GCPs, it will default to WGS84 for the orthophoto output. I believe you'll still need to to use the --odm_georeferencing-useGcp false to use the GPS ephemeris in your images.

seanXYZ, 2015-09-10:
Thank you a lot~ I am already trying. Still running, a little slow

dakotabenjamin, 2015-12-09:
@dbrgn @seanXYZ still having issues? Now a GeoTiff should be created along with a png.

seanXYZ, 2016-01-11:
NO Problem, thanks

dakotabenjamin, 2016-01-11:
Awesome!

closed 2016-01-11 by dakotabenjamin