OpenDroneMap/ODM #146: cmvs returned with code 34816

DarylWM, 2015-08-13:
I haven't found a description for error 34816. I'm guessing 0 points is not a good thing. Any ideas?

  - running cmvs - Thu Aug 13 01:58:20 UTC 2015

Reading bundle...2 cameras -- 0 points in bundle file

2 cameras -- 0 points
Reading images: **
Set widths/heights...done	0 secs
done	0 secs
slimNeighborsSetLinks...done	0 secs
mergeSFM...Floating point exception (core dumped)

quitting cause: 
	"/odm_app/OpenDroneMap/bin/cmvs" pmvs/ 500 2
returned with code 34816

dakotabenjamin, 2015-08-13:
I'm getting the same errors. Which dataset are you using, and what are the specs of your machine?

DarylWM, 2015-08-13:
I'm using my own dataset. I didn't get this problem with the Pacifica dataset, but I haven't tried the other ODM ones yet.

My machine is a:
MacBook Pro (Retina, 13-inch, Late 2013)
2.8 GHz Intel Core i7
16 GB 1600 MHz DDR3
Intel Iris 1536 MB

dakotabenjamin, 2015-08-14:
Hmm. Can you post the bundler log? 

here's my output: 

- running bundler - Wed Aug 12 18:42:56 UTC 2015

[ReadBundleFile] Bundle version: 0.300
[ReadBundleFile] Reading 122 images and 0 points...
[GetJPEGDimensions] File ./IMG_1330.jpg: ( 1600 , 1200 )
[GetJPEGDimensions] File ./IMG_1400.jpg: ( 1600 , 1200 )

@@ Conversion complete, execute "sh pmvs/prep_pmvs.sh" to finalize
@@ (you will first need to edit prep_pmvs.sh to specify your bundler path, 
@@  so that the script knows where to find your
@@  RadialUndistort and Bundle2Vis binaries)
[ReadBundleFile] Bundle version: 0.300
[ReadBundleFile] Reading 122 images and 0 points...
Undistorting image ./IMG_1330.jpg
Undistorting image ./IMG_1400.jpg
[WriteBundleFile] Writing 2 images and 0 points...
[ReadBundleFile] Bundle version: 0.300
[ReadBundleFile] Reading 2 images and 0 points...
Num visible: 0
Num cameras: 2

  - running cmvs - Wed Aug 12 18:43:30 UTC 2015

Reading bundle...2 cameras -- 0 points in bundle file

2 cameras -- 0 points
Reading images: **
Set widths/heights...done       0 secs
done    0 secs
slimNeighborsSetLinks...done    0 secs
mergeSFM...Floating point exception (core dumped)

quitting cause:
        "/code/bin/cmvs" pmvs/ 500 6
returned with code 34816

dakotabenjamin, 2015-08-14:
Looking into the [bundler logs](https://github.com/OpenDroneMap/OpenDroneMap/files/1417/bundlerout.txt), there are a lot of Chierality violations starting at line 40035, then it looks like all the points are being dropped/pruned out (line 43589). I think this problem may stem from f95c49c142c2efcd7782ae83a46bcc3dffabcddd. @chris-cooper do you know anything about this?

DarylWM, 2015-08-15:
The bundler logs don't appear to be in /odm_app/OpenDroneMap/logs. Where else might they be?

dakotabenjamin, 2015-08-15:
Check in your images directory, under
 reconstruction-with-image-size-2400/bundler/out

DarylWM, 2015-08-15:
I too have many Cheirality violations in my bundle/out file.
https://gist.github.com/DarylWM/82aa6a6dec5dc3143c27

dakotabenjamin, 2015-08-18:
0228f671d2 should have fixed this issue. Can you report back on your own dataset?

DarylWM, 2015-08-19:
I deleted the previous reconstruction directories, did a git pull, executed
install.sh, and ran it on my dataset.

  - running cmvs - Wed Aug 19 00:57:20 UTC 2015

Reading bundle...3 cameras -- 4 points in bundle file

****

3 cameras -- 4 points

Reading images: ***

Set widths/heights...done 0 secs

done 0 secs

slimNeighborsSetLinks...done 0 secs

mergeSFM...Floating point exception (core dumped)

quitting cause:

"/odm_app/OpenDroneMap/bin/cmvs" pmvs/ 500 2

returned with code 34816

Did I do it wrong?

DarylWM, 2015-08-19:
I deleted the previous reconstruction directories, did a git pull, executed install.sh, and ran it on my dataset. Still getting the 34816 error. The bundle step seemed to have fewer errors this time though.
https://gist.github.com/DarylWM/31b18acdb3fe93565440

  - running cmvs - Wed Aug 19 00:57:20 UTC 2015

Reading bundle...3 cameras -- 4 points in bundle file
****
3 cameras -- 4 points
Reading images: ***
Set widths/heights...done	0 secs
done	0 secs
slimNeighborsSetLinks...done	0 secs
mergeSFM...Floating point exception (core dumped)

quitting cause: 
	"/odm_app/OpenDroneMap/bin/cmvs" pmvs/ 500 2
returned with code 34816

dakotabenjamin, 2015-08-19:
So this is a different error. It could be a data problem... Could you give
some info about the data you are working with? Jhead output would help,
too.

DarylWM, 2015-08-19:
I uploaded the images here:
https://dl.dropboxusercontent.com/u/16780642/data.zip

It's a small subset of the dataset I'm wanting to use, but it's enough to reproduce the error.

Can you point me to where I can find the jhead output?

DarylWM, 2015-08-19:
Sorry, just found jhead. I've been using exiftools. Here's the jhead output:

File name    : image_1434334225-rgb.jpg
File size    : 470011 bytes
File date    : 2015:06:15 12:10:25
Camera make  : RaspberryPi
Camera model : RP_OV5647
Date/Time    : 2015:06:15 12:10:25
Resolution   : 2592 x 1944
Flash used   : No
Focal length :  3.6mm
Exposure time: 0.0020 s  (1/504)
Aperture     : f/2.9
ISO equiv.   : 400
Whitebalance : Auto
Metering Mode: center weight
Exposure     : aperture priority (semi-auto)
GPS Latitude : S 37d 57m 18.137s
GPS Longitude: E 145d 14m 10.824s
GPS Altitude :  45.00m
JPEG Quality : 75
Comment      : Roll:-28.99,Pitch:-12.56,Heading:21.24,RelAltitude:29.67

dakotabenjamin, 2015-08-19:
Are you also only running ODM with these 8 photos? Have you tried using more photos? These look good, and have a lot of overlap. I'm impressed that you're using Raspberry Pi.

DarylWM, 2015-08-19:
When I put these (11) images through ODM I get the "mergeSFM...Floating point exception" problem. These images are just a subset of the full set. Running the full set was when I first encountered the problem.

Is there too much overlap?

dakotabenjamin, 2015-08-20:
Overlap is generally a good thing, I don't think you can have have too much, although it may have an effect on processing time.

In your bundler log, it looks like it is removing all but 4 points in 3 photos, which isn't good. I ran the photos on my side, got through pmvs, but failed at meshing (because there are too few photos to create a mesh). [Here is my bundler out file.](https://gist.github.com/dakotabenjamin/fc5751bd9edf82074521) You can see that not as many points were removed at the end and all 11 cameras were retained. This was on a fresh pull/install on a native ubuntu machine.

closed 2016-10-12 by dakotabenjamin