OpenDroneMap/ODM #147: Bundler quitted with error code 34304

merkato, 2015-08-15:
I've reinstalled ODM after new improvements, and there is problem. With '-use-ceres' I've getting this error:

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0815 05:52:01.134687  3692 reorder_program.cc:253] Check failed: size_of_first_elimination_group >= 1 (0 vs. 1) Congratulations, you found a Ceres bug! Please report this error to the developers.
*** Check failure stack trace: ***
Aborted (core dumped)

quitting cause: 
	"/home/mechanik/dev/OpenDroneMap/bin/bundler" "/home/mechanik/dev/odm_data/boruszyn/reconstruction-with-image-size-2400/list.txt" --options_file "/home/mechanik/dev/odm_data/boruszyn/reconstruction-with-image-size-2400/options.txt" > bundle/out
returned with code 34304

If I comment out this parameter line in run.pl everything is going well - reconstruction ends with orthophoto. I've tried boruszyn and benchmark datasets.

Is it known Ceres bug, or I should report this?

With regards, Tomasz

merkato, 2015-08-15:
Last few lines from bundle.out:

  Adding match 57887 ==> 58309 -- -664.654 227.935 ==> -235.518 91.933 [1440]
 tri.error[1633] = 3.636
[DumpOutputFile] Wrote file in 0.009s
[CheckPointKeyConsistency] There were 0 errors
[Ceres] num_points = 1441
[Ceres] num_cameras = 2
[Ceres] num_params = 4341
[Ceres] num_projections = 2882
[Ceres] Using CHOLMOD

smathermather, 2015-08-17:
I'd say report it, @merkato.

dakotabenjamin, 2015-08-18:
0228f671d2 should have fixed this.

closed 2015-08-18 by dakotabenjamin