OpenDroneMap/ODM #150: First-time installation: odm_vagrant

rionlerm, 2015-08-19:
​I am installing OpenDroneMap on a Windows host for the first time. So far so good, but the installation seems to be pausing for hours at command line:

> clapack
- building clapack

​What could the issue be?
I use Oracle VM, Vagrant machine (odm_vagrant), Ubuntu (odm vagrant explains). Everything works fine when I run the commands through Git Shell, but after doing all the downloads and unpacking the files I suppose it gets stuck as mentioned above.

My org. is currently in process purchasing a UAV and I need ODM to run successfully as we do not want to purchase Pix4D or PhotoScan for image compilation, analysis and point clouds.

dakotabenjamin, 2015-08-19:
Does the install script eventually finish?

rionlerm, 2015-08-19:
No!

*Rion Lerm (jnr.)*
(013) 735 3536
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)
Click here to learn about Rion's responsibilities within SAEON
<http://www.saeon.ac.za/enewsletter/archives/2014/october2014/doc09>
Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2015-08-19:
I answered on the email list: http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users 

Sorry for the delayed response there.

closed 2015-08-19 by dakotabenjamin