OpenDroneMap/ODM #154: Python Port

dakotabenjamin, 2015-08-20:
Hey all,

There is a new branch called python-port. This is a big leap forward that should resolve a few issues, but it needs plenty of testing before we switch over completely. 

These are the changes so far:

 * A new Python script: run.py
       * Directly translated from latest run.pl
       * Old run.pl has not been removed
 * Accelerated the key point matching step when image EXIF data is available
       * Performed by measuring distances between all images.
       * Operates by matching either towards the k-nearest neighbors or against all images closer than a provided threshold.
 * Put GPS positions in EXIF based on GCP
 * Created a geo-referenced point cloud as a LAS file
 * Created a geo-referenced orthophoto as a GeoTIFF file

Known problems:

 * install.sh sometimes hangs when fetching LAStools.zip, not sure why. Maybe a general curl problem that has occurred before?

Big thanks to the Spotscale folks for providing the code!

merkato, 2015-08-28:
"Put GPS positions in EXIF based on GCP"
If I correctly understand this, and code, right now it should be possible to create orthophoto from my dataset with GCP?

I've tried Boruszyn dataset with gcp_list.txt and my output is like this:

Script generated ../odm_data/boruszyn/reconstruction-with-image-size-2400-results/option-0000_georef.ply correctly, but for GeoTIFF and LAS there is no output.

  - running texturing - Fri Aug 28 13:22:00  2015

  - running georeferencing - Fri Aug 28 13:44:08  2015
Error in OdmExtractUtm:
No valid GPS position found
For more detailed information, see log file.

quitting cause: 
	"/home/mechanik/dev/OpenDroneMap/bin/odm_extract_utm" -imagesPath /home/mechanik/dev/odm_data/boruszyn/ -imageListFile /home/mechanik/dev/odm_data/boruszyn/reconstruction-with-image-size-2400/pmvs/list.rd.txt -outputCoordFile /home/mechanik/dev/odm_data/boruszyn/reconstruction-with-image-size-2400/odm_georeferencing/coordFile.txt
returned with code 256.

What I'm doing wrong? ;)

dakotabenjamin, 2015-08-28:
Did you run with the tag `--odm_georeferencing-useGcp True`?

dakotabenjamin, 2015-08-28:
I'm getting the same error too, also during keypoint matching.

merkato, 2015-08-28:
Hmm.. I've rerun  with this tag, and:

 - running orthophoto generation - Fri Aug 28 14:05:10  2015
    Warning: No geo-referenced orthophoto created due to missing geo-referencing or corner coordinates.

Restarted whole process, but:

  - matching keypoints - Fri Aug 28 14:07:48  2015
Error when performing utm_extraction, dumping log file:
Error in OdmExtractUtm:
No valid GPS position found
For more detailed information, see log file.

Maybe problem lays in coordinates of GCP? They're in local polish geodetic system (EPSG:2180), not in geographic  ?

dakotabenjamin, 2015-08-28:
```
imagesPath was set to: /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400
imageListFile was set to: image_list_tmp.txt
outputCoordFile was set to: utm_coord_tmp.txt
Error in OdmExtractUtm:
No valid GPS position found
```
Above is my odm_extracting_utm_log.txt. This error is thrown on line 318 of odm_extract_utm/src/UtmExtractor.cpp

It looks like that function, UtmExtractor::parsePosition, only looks at the jhead output to parse position, and the boruszyn data has no ephemeris, so it fails there.

dakotabenjamin, 2015-08-28:
I think the thing to do here is to write a condition to only run the knn-preselect if gps data exists in the exif. For now, you should run with `--matcher-preselect False`

darkpioneer, 2015-08-28:
Ran command ../../run.py --odm_georeferencing-useGcp False --start-with odm_georeferencing
received error.

This is on the benchmark sample set useing a fresh ec2 r3.8xlarge instance.

  - running georeferencing - Fri Aug 28 22:26:19  2015
Warning: No GCP file. Consider rerunning with argument --odm_georeferencing-useGcp false --start-with odm_georeferencing
Skipping orthophoto
    Creating geo-referenced LAS file (expecting warning)...
ERROR: cannot open file '/run/shm/OpenDroneMap/odm_data/benchmark/reconstruction-with-image-size-2400-results/option-0000_georef.ply'
ERROR: cannot open lasreadertxt with file name '/run/shm/OpenDroneMap/odm_data/benchmark/reconstruction-with-image-size-2400-results/option-0000_georef.ply'
ERROR: could not open lasreader

quitting cause: 
	"/run/shm/OpenDroneMap/bin/txt2las" -i /run/shm/OpenDroneMap/odm_data/benchmark/reconstruction-with-image-size-2400-results/option-0000_georef.ply -o /run/shm/OpenDroneMap/odm_data/benchmark/reconstruction-with-image-size-2400-results/pointcloud_georef.laz -skip 30 -parse xyzRGBssss -set_scale 0.01 0.01 0.01 -set_offset 440138 4588397 0 -translate_xyz 440138 4588397 0 -epsg 32617
returned with code 256.

smathermather, 2015-08-28:
I really should remove the benchmark dataset -- it is has poor geometry and rarely resolves well.  I'll be posting some new and decent datasets in the next few weeks. In the mean time, a better dataset is Seneca.

smathermather, 2015-08-28:
"Could not open lasreader" -- did the install successfully complete after you pulled down the latest changes?

darkpioneer, 2015-08-29:
It looks like the issue is due to local-gen on the ec2 instance, will install and try again
"locale::facet::_S_create_c_locale name not valid"

darkpioneer, 2015-08-29:
Tryed again, this time setting export LC_ALL="en_US.UTF-8" and run the seneca dataset
Finished with no errors but this is the odm_orthphoto.png
[image]

smathermather, 2015-08-29:
@dakotabenjamin had a similar result on an earlier run -- a subset of the total area. He didn't see it later when he re-ran, so something strange is going on. Any thoughts, Dakota?

darkpioneer, 2015-08-29:
I tryed a re-run but had the same result. Probably related to issue #115
In the pvsm folder the list.rd.txt only contains the 4 images, same as his issue

./IMG_0512.jpg
./IMG_0513.jpg
./IMG_0544.jpg
./IMG_0596.jpg

dakotabenjamin, 2015-08-29:
I have a hunch that the problem lies in the Bundle2PMVS and Undistort
processes run after bundler, but I haven't been able to isolate further.

smathermather, 2015-08-31:
Should cc @petrasovaa, on this, as this branch has LAS output of georeferenced point cloud, a feature she requested.

dakotabenjamin, 2015-08-31:
I don't think this is a factor, but when I ran the Seneca set successfully, I had done it on a native ubuntu machine (not a VM), also using --resize-to 1200

dakotabenjamin, 2015-08-31:
@merkato and @darkpioneer could you open up new issues for these and tag them 'python'?

julianandres, 2015-09-14:
Hello, 
I install with python branch but install.sh is deprecated, this install ceres-solver and google-glog don't exist

seanXYZ, 2015-09-15:
me too @julianandres

dakotabenjamin, 2015-12-09:
Closing as the python-port has been merged. if you have any more issues, please open a new one.

closed 2015-12-09 by dakotabenjamin