OpenDroneMap/ODM #155: [python-port] Bundler is only producing a point cloud of a subset of photos

dakotabenjamin, 2015-08-20:
This problem seems to be happening in the bundler, maybe before. see log after successfully completing on odm_data_seneca:
        [RunSFM] outlier removal took 0.004s
        [RunSFM] Removing 0 outliers
        [RemoveBadPointsAndCameras] Pruned 0 points
          focal lengths:
           [000] 1721.472 (2496.774) ./IMG_0512.jpg 66; -2.984e-02 9.750e-03
           [001] 1722.760 (2496.774) ./IMG_0596.jpg 150; -3.418e-02 1.782e-02
           [002] 1704.304 (2496.774) ./IMG_0513.jpg 67; -3.164e-02 1.611e-02
           [003] 1779.096 (2496.774) ./IMG_0544.jpg 98; -3.597e-02 2.266e-02
        [SifterApp::DumpOutputFile] Wrote file in 0.034s
        Error: parent not found
        [SifterApp::BundleAdjust] max_matches = 0
        [BundleAdjust] Bundle adjustment took 188.722s
        [SifterApp::DumpOutputFile] Wrote file in 0.037s
        [BaseApp::SetMatchesFromPoints] Setting up matches...
        [BaseApp::SetMatchesFromPoints] Done!
        [FixReflectionBug] Reflecting scene...

        Those 4 photos show up in pmvs/visualize as well, and appear to be the extent of the point cloud/mesh:
The old orthophoto output:
[image]

new orthophoto output:
[image]

Also, this is probably related to #115

smathermather, 2015-08-24:
I'm wondering if this is a different problem altogether, and is a function of the new optimization for selecting image pairs. If this step is pre-filtering too strongly, only to eliminate the necessary number of pairs, this would explain the new behavior. Re-running with this functionality turned off or turned up would verify my hunch.

dakotabenjamin, 2015-08-24:
Well, I ran seneca again and got a full reconstruction, so I've yet to pin down the issue. Here's the [bundler out](https://gist.github.com/dakotabenjamin/28f1cce235fa7fd2bbc6) file from my successful run. Compare to the [bundler out](https://gist.github.com/dakotabenjamin/4534c1cfbd4ae5b710a3) that created the subsetted results above.

dakotabenjamin, 2015-09-02:
This issue arises specifically when run without the resize tag or when --resize-to 2400. I tried at 800, 1200, and 2048 with success, but 2400 produces the subset above.

dakotabenjamin, 2015-09-02:
After running some tests, I've concluded that this is probably just a data issue.The section being subset is an area with the most overlap and the least need for image correction. I've also not found this issue with other datasets.

closed 2015-09-02 by dakotabenjamin