OpenDroneMap/ODM #166: Georeferencing in run.py

hwiyoung, 2015-09-04:
I'm using my dataset with GPS coordinates in EXIF. When I did run.pl, it didn't show any error messages and produced right ouput 'odm_orthphoto.png'. But when I do run.py, it show error messages like this
[image]

But I have a inputCoordFile 'coordfile.txt', which is same with 'coordfile.txt' produced when I did run.pl.
I don't know what is the problem.

dakotabenjamin, 2015-09-04:
That's weird. I thought it might be a typo in run.py, but -inputCoordFile is the argument needed in odm_georef (check for yourself, line 336 in Georef.cpp). 

I've never seen that 'Purpose' text before, can you send more of the std output?

hwiyoung, 2015-09-07:
What is the std output you mean? Is the output file I can get when I did run.pl?

dakotabenjamin, 2015-09-08:
try rerunning run.py with this on the end on the line: `>stdout.txt` then when it finishes or fails, post that text file.

hwiyoung, 2015-09-09:
I tried run.py as you told and it is the result.

>  - configuration:
{'start_with': 'odm_georeferencing', 'end_with': 'odm_orthophoto', 'force_focal': None, 'odm_texturing_textureWithSize': 3600, 'matcher_threshold': 2.0, 'cmvs_maxImages': 500, 'zip_results': False, 'pmvs_threshold': 0.7, 'matcher_preselect': True, 'resize_to': 2400, 'odm_meshing_solverDivide': 9, 'pmvs_wsize': 7, 'odm_texturing_textureResolution': 4096, 'matcher_ratio': 0.6, 'odm_meshing_samplesPerNode': 1, 'odm_meshing_maxVertexCount': 100000, 'pmvs_level': 1, 'run_only': None, 'odm_meshing_octreeDepth': 9, 'pmvs_csize': 2, 'matcher_kDistance': 20, 'odm_georeferencing_useGcp': False, 'odm_georeferencing_gcpFile': 'gcp_list.txt', 'matcher_useKnn': True, 'pmvs_minImageNum': 3, 'force_ccd': None}

>  - source files - Wed Sep 09 01:24:40  2015
     using 140304_000062.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000063.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000064.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000065.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000066.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000067.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000068.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000069.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000070.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000071.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000072.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000073.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000074.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000075.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000076.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000077.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000078.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000079.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000080.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000081.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000082.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000083.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000084.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000085.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000086.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000087.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000088.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000089.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000090.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000091.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000092.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000093.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000094.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000095.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000096.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000097.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000098.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000099.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000100.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using Georef.exe

>Purpose:
Georeference a textured mesh with the use of ground control points or exif data from the images.
Usage:
The program requires a path to a camera bundle file, a camera georeference coords file, and an input OBJ mesh file. All other input parameters are optional.

>The following flags are available
Call the program with flag "-help", or without parameters to print this message, or check any generated log file.
Call the program with flag "-verbose", to print log messages in the standard output stream as well as in the log file.

>Parameters are specified as: "-<argument name> <argument>", (without <>), and the following parameters are configureable: 
"-bundleFile <path>" (mandatory)
"Input cameras bundle file.

>"-gcpFile <path>" (mandatory if using ground control points)
Path to the file containing the ground control points used for georeferencing.
The file needs to be on the following line format:
easting northing height pixelrow pixelcol imagename

>"-coordFile <path>" (mandatory if using exif data)
"Input cameras geroreferenced coords file.

>"-inputFile <path>" (mandatory)
"Input obj file that must contain a textured mesh.
[... 41 lines cut ...]
     using 140304_000127.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
[... cut ...]
     using 140304_000140.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000141.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000142.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm
     using 140304_000143.JPG     dimensions: 4912x3264 / focal: 24.0mm / ccd: 23.4mm

>    found 82 usable images

>  using max image size of 2400 x 2400

>  - running georeferencing - Wed Sep 09 01:24:42  2015

And I tried run.py with caliterra dataset, but the result is same.

hwiyoung, 2015-09-30:
I reinstall OpenDroneMap and the problem was solved. It might have been a conflicth between python-port and gh-pages.

closed 2015-09-30 by hwiyoung